AlienWolfX/UZ801-USB-MODEM #3: cant use rndis or usb tethering

mchasby, 2024-07-07:
cant use internet via usb, any solutions?

AlienWolfX, 2024-07-07:
Are you using the stock firmware?

mchasby, 2024-07-07:
No, I'm using this firmware https://github.com/AlienWolfX/UZ801-USB_MODEM/blob/main/files/openwrt-UZ801_v3.2.tar.gz

and flash (fsc.bin, fsg.bin, modemst1.bin, modemst2.bin) from my stock dump

AlienWolfX, 2024-07-07:
I see, Are you using windows? If so, try installing this driver [RNDIS DRIVER](https://github.com/milkv-duo/duo-files/raw/main/common/RNDIS_drivers_20231018.zip)

mchasby, 2024-07-08:
okay thanks for windows the driver worked very well.
but for mikrotik routeros, it cant detect lte interfaces from uz801 modem

AlienWolfX, 2024-07-08:
> okay thanks for windows the driver worked very well. but for mikrotik routeros, it cant detect lte interfaces from uz801 modem

No worries. Regarding Mikrotik and the UZ801, I think the UZ801 is not supported by RouterOS.

closed 2024-07-09 by AlienWolfX