AlienWolfX/UZ801-USB-MODEM #6: Openwrt Image

xMase, 2024-07-25:
Hi thanks in advance for your work with openwrt it's amazing. 
How do you compile the openwrt firmware for this device? 
i need it just in case i want to upgrade the version.

AlienWolfX, 2024-07-26:
Hello, I didn't compiled the openwrt firmware for this device but I got it from here [OpenWRT Images](https://blog.mxdyeah.top/mxdyeah_blog_post/41.html)

danyaPostfactum, 2024-10-27:
I have installed this image. Unfortunately, wifi speed is limited to 6.5Mbit/s

[image]
I tried different wifi settings, but no luck…

AlienWolfX, 2024-10-27:
> I have installed this image. Unfortunately, wifi speed is limited to 6.5Mbit/s

Did you try installing this?
[kmod-wcn36xx.zip](https://github.com/user-attachments/files/17534794/kmod-wcn36xx.zip)