AlienWolfX/UZ801-USB-MODEM #7: Debian Image

voltrare, 2024-07-30:
hi, the debian image link is down, do you have the file?

AlienWolfX, 2024-07-30:
Hello, I've created a temporary link [here](https://drive.google.com/file/d/1mqujlplGCXK3hC5EQKRqMvNC1FBe3loh/view?usp=sharing)

voltrare, 2024-07-30:
thanks so much i have v3.2 variant do i need to other stuff on it?

AlienWolfX, 2024-07-30:
> thanks so much i have v3.2 variant do i need to flash other stuff on it? 
> 

No, you don't have to.

closed 2024-07-30 by voltrare

danyaPostfactum, 2024-10-28:
> Download the file using: wget https://download.wvthoog.nlopenstick-uz801-v3.0.zip.

Misspelled link. It should be https://download.wvthoog.nl/openstick-uz801-v3.0.zip

danyaPostfactum, 2024-10-31:
Do you have  Debian (Based on 5.15 msm8916 mainline) file?
The link http://www-cs-students.stanford.edu/~kaan/test.html is down.