AlienWolfX/UZ801-USB-MODEM #8: EDL Loader ?

alexandrglm, 2024-10-19:


alexandrglm, 2024-10-19:
Hi.
First of all, I appreciate your contribution. It's really **IMPRESSIVE** !
I would like to ask you about EDL. What loader are you using?
Is it able to read main memory?

Here is a log of the problems I'm having:

`PS C:\swap\UZ801\git\edl-master> python3 edl printgpt
Qualcomm Sahara / Firehose Client V3.62 (c) B.Kerler 2018-2024.
main - Trying with no loader given ...
main - Waiting for the device
main - Device detected :)
sahara - Protocol version: 2, Version supported: 1
main - Mode detected: sahara
sahara -
Version 0x2
------------------------
HWID:              0x007050e100000000 (MSM_ID:0x007050e1,OEM_ID:0x0000,MODEL_ID:0x0000)
CPU detected:      "MSM8916"
PK_HASH:           0xcc3153a80293939b90d02d3bf8b23e0292e452fef662c74998421adad42a380f
Serial:            0x00415511

sahara - Possibly unfused device detected, so any loader should be fine...
sahara - Possible loader available: C:\swap\UZ801\git\edl-master\edlclient\..\Loaders\cyanogen\007050e100000000_4614048173062ae4_fhprg_peek.bin
sahara - Possible loader available: C:\swap\UZ801\git\edl-master\edlclient\..\Loaders\cyanogen\007050e100000000_4e3eefa63a67eb7a_fhprg_peek.bin
sahara - Possible loader available: C:\swap\UZ801\git\edl-master\edlclient\..\Loaders\cyanogen\007050e100000000_d36c6073c9c2cb1c_fhprg_peek.bin
sahara - Possible loader available: C:\swap\UZ801\git\edl-master\edlclient\..\Loaders\lenovo_motorola\007050e100000000_99c8c13e374c34d8_fhprg_peek.bin
sahara - Possible loader available: C:\swap\UZ801\git\edl-master\edlclient\..\Loaders\longcheer\007050e100000000_3022817d373fd7f9_fhprg_peek.bin
sahara - Possible loader available: C:\swap\UZ801\git\edl-master\edlclient\..\Loaders\lyf\007050e100000000_394a2e47cf830150_fhprg_peek.bin
sahara - Possible loader available: C:\swap\UZ801\git\edl-master\edlclient\..\Loaders\qualcomm\factory\msm8916\007050e100000000_8ecf3eaa03f772e2_fhprg_peek.bin
sahara - Possible loader available: C:\swap\UZ801\git\edl-master\edlclient\..\Loaders\xiaomi\007050e100000000_278448179ac756a1_fhprg_peek.bin
sahara - Possible loader available: C:\swap\UZ801\git\edl-master\edlclient\..\Loaders\xiaomi\007050e100000000_50838757eab7c632_fhprg_peek_wt88047.bin
sahara - Trying loader: C:\swap\UZ801\git\edl-master\edlclient\..\Loaders\cyanogen\007050e100000000_4614048173062ae4_fhprg_peek.bin
sahara - Protocol version: 2, Version supported: 1
sahara - Uploading loader C:\swap\UZ801\git\edl-master\edlclient\..\Loaders\cyanogen\007050e100000000_4614048173062ae4_fhprg_peek.bin ...
sahara - 32-Bit mode detected.
sahara - Firehose mode detected, uploading...
sahara - Loader successfully uploaded.
main - Trying to connect to firehose loader ...
firehose_client
firehose_client - [LIB]: No --memory option set, we assume "eMMC" as default ..., if it fails, try using "--memory" with "UFS","NAND" or "spinor" instead !
firehose
firehose - [LIB]: !DEBUG! rsp.data: 'bytearray(b'<?xml version="1.0" encoding="UTF-8" ?><data><log value="Host\'s payload to target size is too large" /></data><?xml version="1.0" encoding="UTF-8" ?><data><log value="logbuf@0x0801CDB8 fh@0x08019C20" /></data><?xml version="1.0" encoding="UTF-8" ?><data><response value="NA')'
firehose
firehose - [LIB]: !DEBUG! rsp.data: 'bytearray(b'K" MinVersionSupported="1" MemoryName="eMMC" MaxPayloadSizeFromTargetInBytes="4096" MaxPayloadSizeToTargetInBytes="16384" MaxPayloadSizeToTargetInBytesSupported="16384" MaxXMLSizeInBytes="4096" Version="1" TargetName="8916" /></data><?xml version="1.0" encoding="UTF-8" ?><data><log value="logbuf@0x0801CDB8 fh@0x08019C20" /></data>')'
firehose
firehose - [LIB]: !DEBUG! rsp.data: 'bytearray(b'<?xml version="1.0" encoding="UTF-8" ?><data><log value="Host\'s payload to target size is too large" /></data><?xml version="1.0" encoding="UTF-8" ?><data><log value="logbuf@0x0801CDB8 fh@0x08019C20" /></data><?xml version="1.0" encoding="UTF-8" ?><data><response value="NA')'
firehose - TargetName=
firehose - MemoryName=eMMC
firehose - Version=
firehose - Trying to read first storage sector...
Traceback (most recent call last):
  File "C:\swap\UZ801\git\edl-master\edlclient\Library\firehose.py", line 888, in configure
    if "MemoryName" in rsp.data:
       ^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: a bytes-like object is required, not 'str'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\swap\UZ801\git\edl-master\edlclient\Library\firehose.py", line 888, in configure
    if "MemoryName" in rsp.data:
       ^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: a bytes-like object is required, not 'str'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\swap\UZ801\git\edl-master\edlclient\Library\firehose.py", line 888, in configure
    if "MemoryName" in rsp.data:
[... 7 lines cut ...]
    base.run()
  File "C:\swap\UZ801\git\edl-master\edl", line 384, in run
    if fh.connect(sahara):
       ^^^^^^^^^^^^^^^^^^
  File "C:\swap\UZ801\git\edl-master\edlclient\Library\firehose_client.py", line 114, in connect
    if self.firehose.configure(0):
       ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\swap\UZ801\git\edl-master\edlclient\Library\firehose.py", line 892, in configure
    return self.configure(lvl + 1)
           ^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\swap\UZ801\git\edl-master\edlclient\Library\firehose.py", line 892, in configure
    return self.configure(lvl + 1)
           ^^^^^^^^^^^^^^^^^^^^^^^
[... cut ...] gure
    return self.configure(lvl + 1)
           ^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\swap\UZ801\git\edl-master\edlclient\Library\firehose.py", line 1013, in configure
    rsp = self.cmd_read_buffer(0, 1, 1, False)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\swap\UZ801\git\edl-master\edlclient\Library\firehose.py", line 743, in cmd_read_buffer
    return response(resp=False, data=resData, error=rsp[2])
                                                    ~~~^^^
KeyError: 2`

Thanks.

AlienWolfX, 2024-10-19:
Hello, are you using EDL under Windows? If so, I recommend using Linux (if you have one available) as EDL under Windows tends to produce errors with this device. Here's a full log of mine while running the command woking loader is `007050e100000000_8ecf3eaa03f772e2_fhprg_peek.bin`:

```
super@yskaela:~/edl$ python3 edl printgpt
Qualcomm Sahara / Firehose Client V3.62 (c) B.Kerler 2018-2024.
main - Trying with no loader given ...
main - Waiting for the device
main - Device detected :)
sahara - Protocol version: 2, Version supported: 1
main - Mode detected: sahara
sahara - 
Version 0x2

HWID:              0x007050e100000000 (MSM_ID:0x007050e1,OEM_ID:0x0000,MODEL_ID:0x0000)
CPU detected:      "MSM8916"
PK_HASH:           0xcc3153a80293939b90d02d3bf8b23e0292e452fef662c74998421adad42a380f
Serial:            0x0575346f

sahara - Possibly unfused device detected, so any loader should be fine...
sahara - Possible loader available: /home/super/edl/edlclient/../Loaders/qualcomm/factory/msm8916/007050e100000000_8ecf3eaa03f772e2_fhprg_peek.bin
sahara - Possible loader available: /home/super/edl/edlclient/../Loaders/qualcomm/factory/msm8916/007050e100000000_394a2e47cf830150_fhprg_peek.bin
sahara - Possible loader available: /home/super/edl/edlclient/../Loaders/cyanogen/007050e100000000_d36c6073c9c2cb1c_fhprg_peek.bin
sahara - Possible loader available: /home/super/edl/edlclient/../Loaders/cyanogen/007050e100000000_4614048173062ae4_fhprg_peek.bin
sahara - Possible loader available: /home/super/edl/edlclient/../Loaders/cyanogen/007050e100000000_4e3eefa63a67eb7a_fhprg_peek.bin
sahara - Possible loader available: /home/super/edl/edlclient/../Loaders/xiaomi/007050e100000000_278448179ac756a1_fhprg_peek.bin
sahara - Possible loader available: /home/super/edl/edlclient/../Loaders/xiaomi/007050e100000000_50838757eab7c632_fhprg_peek_wt88047.bin
sahara - Possible loader available: /home/super/edl/edlclient/../Loaders/longcheer/007050e100000000_3022817d373fd7f9_fhprg_peek.bin
sahara - Possible loader available: /home/super/edl/edlclient/../Loaders/lenovo_motorola/007050e100000000_99c8c13e374c34d8_fhprg_peek.bin
sahara - Trying loader: /home/super/edl/edlclient/../Loaders/qualcomm/factory/msm8916/007050e100000000_8ecf3eaa03f772e2_fhprg_peek.bin
sahara - Protocol version: 2, Version supported: 1
sahara - Uploading loader /home/super/edl/edlclient/../Loaders/qualcomm/factory/msm8916/007050e100000000_8ecf3eaa03f772e2_fhprg_peek.bin ...
sahara - 32-Bit mode detected.
sahara - Firehose mode detected, uploading...
sahara - Loader successfully uploaded.
main - Trying to connect to firehose loader ...
firehose_client
firehose_client - [LIB]: No --memory option set, we assume "eMMC" as default ..., if it fails, try using "--memory" with "UFS","NAND" or "spinor" instead !
firehose - TargetName=
firehose - MemoryName=eMMC
firehose - Version=
firehose - Trying to read first storage sector...
firehose - Running configure...
firehose
firehose - [LIB]: GetStorageInfo command isn't supported.
firehose_client - Supported functions:

Parsing Lun 0:

GPT Table:

modem:               Offset 0x0000000004000000, Length 0x0000000004000000, Flags 0x1000000000000000, UUID c656e4f7-ed17-6e01-239c-9dd35cebae17, Type EFI_BASIC_DATA, Active False
sbl1:                Offset 0x0000000008000000, Length 0x0000000000080000, Flags 0x0000000000000000, UUID e920226e-6d6e-3c93-9a43-ba3e676767ad, Type 0xdea0ba2c, Active False
sbl1bak:             Offset 0x0000000008080000, Length 0x0000000000080000, Flags 0x0000000000000000, UUID f1b071eb-9086-0b2b-a209-473697ac5590, Type EFI_BASIC_DATA, Active False
aboot:               Offset 0x0000000008100000, Length 0x0000000000100000, Flags 0x0000000000000000, UUID a00e22fe-c273-7d77-9931-42bb796d5d42, Type 0x400ffdcd, Active False
abootbak:            Offset 0x0000000008200000, Length 0x0000000000100000, Flags 0x0000000000000000, UUID 73842151-9df2-0d3d-5992-16d080582cd1, Type EFI_BASIC_DATA, Active False
rpm:                 Offset 0x0000000008300000, Length 0x0000000000080000, Flags 0x0000000000000000, UUID 7bc3e465-deb1-8599-f8ce-7e70a18dccf5, Type 0x98df793, Active False
rpmbak:              Offset 0x0000000008380000, Length 0x0000000000080000, Flags 0x0000000000000000, UUID 57ac3008-849e-d0f7-d4f0-783463e31264, Type EFI_BASIC_DATA, Active False
tz:                  Offset 0x0000000008400000, Length 0x0000000000080000, Flags 0x0000000000000000, UUID d71d3e0e-1c17-66c1-9cc2-a890b22f06aa, Type 0xa053aa7f, Active False
tzbak:               Offset 0x0000000008480000, Length 0x0000000000080000, Flags 0x0000000000000000, UUID e2373854-b760-003b-7d8b-4d1632c7c243, Type EFI_BASIC_DATA, Active False
hyp:                 Offset 0x0000000008500000, Length 0x0000000000080000, Flags 0x0000000000000000, UUID 73ea9814-57dd-54f9-508a-734c7740ef7a, Type 0xe1a6a689, Active False
hypbak:              Offset 0x0000000008580000, Length 0x0000000000080000, Flags 0x0000000000000000, UUID 7e78292e-0ff7-1d67-38ea-a98c243a08bd, Type EFI_BASIC_DATA, Active False
pad:                 Offset 0x0000000008600000, Length 0x0000000000100000, Flags 0x0000000000000000, UUID 9810f95f-290d-471f-4863-efe52f28ec2e, Type EFI_BASIC_DATA, Active False
modemst1:            Offset 0x0000000008700000, Length 0x0000000000180000, Flags 0x0000000000000000, UUID 6ed17df0-74d8-e99d-0764-ff5a3f3dbd11, Type 0xebbeadaf, Active False
[... cut ...]
cache:               Offset 0x000000004118c000, Length 0x0000000008000000, Flags 0x1000000000000000, UUID d271393d-d43c-0e34-3c04-65f3b960e910, Type EFI_BASIC_DATA, Active False
recovery:            Offset 0x000000004918c000, Length 0x0000000001000000, Flags 0x1000000000000000, UUID da35c838-dde0-e1ad-04d2-f9fb0c87a021, Type 0x20117f86, Active False
userdata:            Offset 0x000000004a18c000, Length 0x000000009ce6fe00, Flags 0x1000000000000000, UUID 1052770c-9294-6c13-1703-8359023c077e, Type EFI_BASIC_DATA, Active False

Total disk size:0x00000000e7000000, sectors:0x0000000000738000
```

alexandrglm, 2024-10-20:
Hello,

I've attached a Windows log after several attempts to get EDL working on
Debian. Unfortunately, I haven't had any success. I've tried using a
bootable USB with Debian 22.04 and EDL, but it doesn't work. I also tried
compiling the source code, but I encountered dependency issues with pylzma,
which seems to have changed its routines and code. I might have to modify
some of Bkerler's scripts (it's odd that they haven't been updated in two
years), as I'm unable to proceed otherwise.

Thank you for sharing the loader and the log file.

If you're interested in testing on my board, I can provide details about
the chipset, board, firmware, and test scripts, as there are variations
between different manufacturers and models.

Since I haven't been able to start developing for this board due to the
issues mentioned, I've started working on a brute-force search script (in
Python) to find matching folders/files at 192.168.100.1.

Thank you for your support.

Alexandr

AlienWolfX, 2024-10-21:
Hello, I'm interested in the variant you have. Could you please provide details about your chipset, board, firmware, and any test scripts? Also, have you already tried using Miko Service Tools?. Best Regards

alexandrglm, 2024-10-24:
Well, now I can understand why I wasn't able to set an EDL connection...:
My device came already rooted!

It's using another loader from cyanogen loader. I've been able to make a full flash backup (also from every partition, extracting the gpt table, then using ecl to extract every PARTITION_NAME.bin).

Since I still don't know what board/version I have, and even though I would like to finish setting up a functional OpenWRT (with Luci, not just ssh), I'm going to wait without flashing anything so that, even on Android, if you think of any data I can extract for you, let me know.

alexandrglm, 2024-10-24:
From shell, getprop | grep 'ro' :

```
[ro.board.platform]: [msm8916]
[ro.boot.baseband]: [msm]
[ro.boot.bootdevice]: [7824900.sdhci]
[ro.boot.console]: [ttyHSL0]
[ro.boot.emmc]: [true]
[ro.boot.hardware]: [qcom]
[ro.boot.serialno]: /* I removed this from log*/
[ro.bootloader]: [unknown]
[ro.bootmode]: [unknown]
[ro.build.characteristics]: [default]
...
[ro.build.description]: [msm8916_32_512-userdebug 4.4.4 KTU84P eng..20240426 test-keys]
[ro.build.display.id]: [V2.3.11]
[ro.build.fingerprint]: [qcom/msm8916_32_512/msm8916_32_512:4.4.4/KTU84P/eng..20240426:userdebug/test-keys]
[ro.build.host]: [cf4c1a952808]
[ro.build.id]: [UZ801]
[ro.build.product]: [msm8916_32_512]
...
[ro.product.locale.language]: [zh]
[ro.product.locale.region]: [CN]
[ro.product.manufacturer]: [Qualcomm Technology]
...
[ro.product.model]: [UZ801]
[ro.product.name]: [msm8916_32_512]
...
[ro.qualcomm.cabl]: [0]
[ro.revision]: [0]
[ro.ril.svdo]: [false]
[ro.ril.svlte1x]: [false]
...
```

AlienWolfX, 2024-10-24:
> From shell, getprop | grep 'ro' :
>
>
> ```
> [ro.board.platform]: [msm8916]
> [ro.boot.baseband]: [msm]
> [ro.boot.bootdevice]: [7824900.sdhci]
> [ro.boot.console]: [ttyHSL0]
> [ro.boot.emmc]: [true] [...]

Hmm It looks like it's UZ801, did you already tried opening the dongle.?

AlienWolfX, 2024-10-24:
> Well, I can now understand why I wasn't able to set an EDL connection...:
> **My device came already rooted!**
> 
> It's using another loader from cyanogen loader. I've been able to make a full flash backup (also from every partition, extracting the gpt table, then using ecl to extract every PARTITION_NAME.bin).
> 
> Since I still don't know what board/version I have, and even though I would like to finish setting up a functional OpenWRT (with Luci, not just ssh), I'm going to wait without flashing anything so that, even on Android, if you think of any data I can extract for you (android configs, any .img partition, ... ) let me know. 
> 
> Alexandr.
> 
> PS: I can't mirror the screen via adb, idk why ... seems to be "protected" from screenshots, or any other option which prevents from being 'screenshoted' (black .png's).

Did you use adbcontrol?

alexandrglm, 2024-10-24:
- Seems to be the same board Board FY_UZ801_V3.2:
[image]
[image]

- System img (BIN extension, extracted within EDL tool; I also have the full stock backup of 3.6Gb):
[https://www.mediafire.com/file/ijcjdpvz4ixrmcl/UZ801_Stock_System_.bin/file](url)

- Adbcontrol used, already config to my paths, black screens:
[image]

alexandrglm, 2024-10-24:
I have a few questions aside from this. I don't know if this is the place. 
E.g.:
OpenWRT on this stick, does it work the same as on routers? Does it have web access or just ssh?

AlienWolfX, 2024-10-24:
> I have a few questions aside from this. I don't know if this is the place. 
> E.g.:
> 
> - OpenWRT on this stick, does it work the same as on routers? Does it have web access or just ssh?
> 
> - The blue wire soldered to your PCB ... Is it improving LTE coverage? or Wlan?  And the antenna mini connector?

OpenWRT

Yes it works the same as that of routers it also has both ssh and web access tho the only downside is it's buggy

Blue Wire

I used it to bridge the EDL pads on the board incase I bricked my dongle.

AlienWolfX, 2024-10-24:
> - Seems to be the same board Board FY_UZ801_V3.2:
> [image]
> [image]
> 
> 
> - System img (BIN extension, extracted within EDL tool; I also have the full stock backup of 3.6Gb):
> https://www.mediafire.com/file/ijcjdpvz4ixrmcl/UZ801_Stock_System_.bin/file
> 
> 
> - Adbcontrol used, already config to my paths, black screens:
> [image]
> 
> 

Hmm, I see. If you don't mind sending the full stock backup, I’d like to tinker with it and try flashing it on my device since we have the same board version.

alexandrglm, 2024-10-24:
- So the cable makes it sense, GND wire to D+. I'll take this idea.

- About OpenWRT ... Is it so so so buggy? 
My idea for using the dongle is to only use it via USB (RNDIS or CDC Ether), not even it WiFi. 
My current firmware doesn't even allow me to turn off the WiFi network.
So... I was thinking about OpenWRT or Debian or something.............. just because what I am clear about is that I don't want to stay on this "factory" version, which unlike other buyers like you, came already rooted.

- Entire stock dump from mine's here:
https://www.mediafire.com/file/kxhm42dwkbbi4l9/UZ801-FIRMWARE-BACKUP.bin.7z/file

AlienWolfX, 2024-10-24:
> * So the cable makes it sense, GND wire to D+. I'll take this idea.
> * About OpenWRT ... Is it so so so buggy?
>   My idea for using the dongle is to only use it via USB (RNDIS or CDC Ether), not even it WiFi.
>   My current firmware doesn't even allow me to turn off the WiFi network.
>   So... I was thinking about OpenWRT or Debian or something.............. just because what I am clear about is that I don't want to stay on this "factory" version, which unlike other buyers like you, came already rooted.
> * Entire stock dump from mine's here:
>   https://www.mediafire.com/file/kxhm42dwkbbi4l9/UZ801-FIRMWARE-BACKUP.bin.7z/file
> 
> Let me know when you have it downloaded, I don't want to leave copies of my modems, fsc, etc. online.

OpenWRT

It's not that buggy. Sometimes, a DNS error and Routing error pops up suddenly, but if you're good with networking and OpenWRT, it's easy to get around. Also speed with RNDIS is decent. 

I've already downloaded your stock dump

Thanks,

Allen

alexandrglm, 2024-10-24:
Ok. I'm going to test OpenWRT / OpenStick Debian with this dongle.
We tell each other questions and news.

Alexandr.

AlienWolfX, 2024-10-24:
Black Screen with adbcontrol fix:

```bash 
adb shell settings put system screen_off_timeout 2147483647

adb shell input keyevent 26
```

Initial findings:
- newer firmware build (APR 26 2024)

alexandrglm, 2024-10-24:
[image]

It works. It's amazing how old chipsets can work in a new form.

alexandrglm, 2024-10-24:
Model number 
UZ801

Processor info 
Qualcomm Technologies, Inc MSM8916

Android version 
4.4.4

Baseband version 
N958St_Z85_CN_JSXPH1IDN1H213

Kernel version 
3.10.28
user17@cf4c1a952808 #1
Fri Apr 26 17:53:41 CST 2024

Build number 
V2.3.11

[image]

alexandrglm, 2024-10-25:
**OpenWRT img you provide** --> Does not work (my edl refuses to write it .... a strange log problem, I'll log and report it in the future).

**OpenStick/Debian** --> It works.... excepting the Modem LTE.

https://wvthoog.nl/openstick/#4GLTE
They manage to fix the problem by extracting the contents of the modem partition into /lib/firmware.

Mine, for whatever reason, still can't do it.

AlienWolfX, 2024-10-25:
> **OpenWRT** img you provide --> Does not work (my edl refuses to write it)
> 
> ```
> # edl wf wrtWORKABLE.bin 
> Qualcomm Sahara / Firehose Client V3.62 (c) B.Kerler 2018-2024.
> main - Trying with no loader given ...
> main - Waiting for the device
> main - Device detected :)
> sahara - Protocol version: 2, Version supported: 1
> main - Mode detected: sahara
> sahara - 
> Version 0x2
> ------------------------
> HWID:              0x007050e100000000 (MSM_ID:0x007050e1,OEM_ID:0x0000,MODEL_ID:0x0000)
> CPU detected:      "MSM8916"
> PK_HASH:           0xcc3153a80293939b90d02d3bf8b23e0292e452fef662c74998421adad42a380f
> Serial:            0x00415511
> 
> sahara - Possibly unfused device detected, so any loader should be fine...
> sahara - Possible loader available: /usr/local/lib/python3.9/dist-packages/edlclient-3.62-py3.9.egg/edlclient/../Loaders/cyanogen/007050e100000000_4614048173062ae4_fhprg_peek.bin
> sahara - Possible loader available: /usr/local/lib/python3.9/dist-packages/edlclient-3.62-py3.9.egg/edlclient/../Loaders/cyanogen/007050e100000000_4e3eefa63a67eb7a_fhprg_peek.bin
> sahara - Possible loader available: /usr/local/lib/python3.9/dist-packages/edlclient-3.62-py3.9.egg/edlclient/../Loaders/cyanogen/007050e100000000_d36c6073c9c2cb1c_fhprg_peek.bin
> sahara - Possible loader available: /usr/local/lib/python3.9/dist-packages/edlclient-3.62-py3.9.egg/edlclient/../Loaders/lenovo_motorola/007050e100000000_99c8c13e374c34d8_fhprg_peek.bin
> sahara - Possible loader available: /usr/local/lib/python3.9/dist-packages/edlclient-3.62-py3.9.egg/edlclient/../Loaders/qualcomm/factory/msm8916/007050e100000000_394a2e47cf830150_fhprg_peek.bin
> sahara - Possible loader available: /usr/local/lib/python3.9/dist-packages/edlclient-3.62-py3.9.egg/edlclient/../Loaders/qualcomm/factory/msm8916/007050e100000000_8ecf3eaa03f772e2_fhprg_peek.bin
> sahara - Possible loader available: /usr/local/lib/python3.9/dist-packages/edlclient-3.62-py3.9.egg/edlclient/../Loaders/longcheer/007050e100000000_3022817d373fd7f9_fhprg_peek.bin
> sahara - Possible loader available: /usr/local/lib/python3.9/dist-packages/edlclient-3.62-py3.9.egg/edlclient/../Loaders/xiaomi/007050e100000000_278448179ac756a1_fhprg_peek.bin
> sahara - Possible loader available: /usr/local/lib/python3.9/dist-packages/edlclient-3.62-py3.9.egg/edlclient/../Loaders/xiaomi/007050e100000000_50838757eab7c632_fhprg_peek_wt88047.bin
> sahara - Trying loader: /usr/local/lib/python3.9/dist-packages/edlclient-3.62-py3.9.egg/edlclient/../Loaders/cyanogen/007050e100000000_4614048173062ae4_fhprg_peek.bin
> sahara - Protocol version: 2, Version supported: 1
> sahara - Uploading loader /usr/local/lib/python3.9/dist-packages/edlclient-3.62-py3.9.egg/edlclient/../Loaders/cyanogen/007050e100000000_4614048173062ae4_fhprg_peek.bin ...
> sahara - 32-Bit mode detected.
> sahara - Firehose mode detected, uploading...
> sahara - Loader successfully uploaded.
> main - Trying to connect to firehose loader ...
> firehose_client
> firehose_client - [LIB]: No --memory option set, we assume "eMMC" as default ..., if it fails, try using "--memory" with "UFS","NAND" or "spinor" instead !
> firehose
> firehose - [LIB]: !DEBUG! rsp.data: 'bytearray(b'<?xml version="1.0" encoding="UTF-8" ?><data><log value="Host\'s payload to target size is too large" /></data><?xml version="1.0" encoding="UTF-8" ?><data><log value="logbuf@0x0801CDB8 fh@0x08019C20" /></data><?xml version="1.0" encoding="UTF-8" ?><data><response value="NA')'
> firehose
> firehose - [LIB]: !DEBUG! rsp.data: 'bytearray(b'K" MinVersionSupported="1" MemoryName="eMMC" MaxPayloadSizeFromTargetInBytes="4096" MaxPayloadSizeToTargetInBytes="16384" MaxPayloadSizeToTargetInBytesSupported="16384" MaxXMLSizeInBytes="4096" Version="1" TargetName="8916" /></data><?xml version="1.0" encoding="UTF-8" ?><data><log value="logbuf@0x0801CDB8 fh@0x08019C20" /></data>')'
> firehose
> firehose - [LIB]: !DEBUG! rsp.data: 'bytearray(b'<?xml version="1.0" encoding="UTF-8" ?><data><log value="Host\'s payload to target size is too large" /></data><?xml version="1.0" encoding="UTF-8" ?><data><log value="logbuf@0x0801CDB8 fh@0x08019C20" /></data><?xml version="1.0" encoding="UTF-8" ?><data><response value="NA')'
> firehose - TargetName=
> firehose - MemoryName=eMMC
> firehose - Version=
> firehose - Trying to read first storage sector...
> DeviceClass
> DeviceClass - [LIB]: USB Overflow
> DeviceClass
> DeviceClass - [LIB]: USB Overflow
> DeviceClass
> DeviceClass - [LIB]: USB Overflow
> ``` 
> 
> **OpenStick/Debian** --> It works.... excepting the Modem LTE.
> 
> https://wvthoog.nl/openstick/#4GLTE
> They manage to fix the problem by extracting the contents of the modem partition into /lib/firmware.
> 
> Mine, for whatever reason, still can't do it.

What does mmcli -m 0 return?

alexandrglm, 2024-10-25:
```
user@openstick:~$ sudo mmcli -m 0
[sudo] password for user:  
  ----------------------------------
  General  |                   path: /org/freedesktop/ModemManager1/Modem/0
           |              device id: c400ed3331ba4b8c803a2fd5c934b15941d0524a
  ----------------------------------
  Hardware |           manufacturer: 1
           |                  model: 0
           |      firmware revision: UZ801_V01R01B08  1  [Sep 07 2015 23:00:00]
           |         carrier config: default
           |           h/w revision: 10000
           |              supported: gsm-umts, lte
           |                current: gsm-umts, lte
           |           equipment id: 860018046109600
  ----------------------------------
  System   |                 device: qcom-soc
           |                drivers: rpmsg_ctrl, qcom-q6v5-mss, bam-dmux
           |                 plugin: qcom-soc
           |           primary port: wwan0qmi0
           |                  ports: rpmsg_ctrl2 (ignored), wwan0 (net), wwan0at0 (at),  
           |                         wwan0at1 (at), wwan0qmi0 (qmi), wwan1 (net), wwan2 (net), wwan3 (net),  
           |                         wwan4 (net), wwan5 (net), wwan6 (net), wwan7 (net)
  ----------------------------------
  Status   |                   lock: sim-pin2
           |         unlock retries: sim-pin (3), sim-puk (3), sim-pin2 (3), sim-puk2 (3)
           |                  state: disabled
           |            power state: off
  ----------------------------------
  Modes    |              supported: allowed: 3g; preferred: none
           |                         allowed: 4g; preferred: none
           |                         allowed: 3g, 4g; preferred: 4g
           |                         allowed: 3g, 4g; preferred: 3g
           |                current: allowed: 3g, 4g; preferred: 3g
  ----------------------------------
  Bands    |              supported: utran-1, utran-8, eutran-1, eutran-3, eutran-5, eutran-7,  
           |                         eutran-8, eutran-20, eutran-38, eutran-40, eutran-41
           |                current: utran-1, utran-8, eutran-1, eutran-3, eutran-5, eutran-7,  
           |                         eutran-8, eutran-20, eutran-38, eutran-40, eutran-41
  ----------------------------------
  IP       |              supported: ipv4, ipv6, ipv4v6
  ----------------------------------
  3GPP     |                   imei: 860018046109600
           |          enabled locks: fixed-dialing
           |   packet service state: detached
  ----------------------------------
  3GPP EPS |   ue mode of operation: csps-1
           |     initial bearer apn: internet.digimobil.es
           | initial bearer ip type: ipv4
  ----------------------------------
  SIM      |       primary sim path: /org/freedesktop/ModemManager1/SIM/0
           |         sim slot paths: slot 1: /org/freedesktop/ModemManager1/SIM/0 (active)
           |                         slot 2: none
```

alexandrglm, 2024-10-25:
About OpenWRT, don't worry for "When?".
While waiting, i'll try to decompose and debug my dongle with your openwrt img VS my boot aboot modem .....but  ... this weekend.

I'm reflashing stock Android again.

AlienWolfX, 2024-10-25:
> ```
> user@openstick:~$ sudo mmcli -m 0
> [sudo] password for user:  
>   ----------------------------------
>   General  |                   path: /org/freedesktop/ModemManager1/Modem/0
>            |              device id: c400ed3331ba4b8c803a2fd5c934b15941d0524a
>   ----------------------------------
>   Hardware |           manufacturer: 1
>            |                  model: 0
>            |      firmware revision: UZ801_V01R01B08  1  [Sep 07 2015 23:00:00]
>            |         carrier config: default
>            |           h/w revision: 10000
>            |              supported: gsm-umts, lte
>            |                current: gsm-umts, lte
>            |           equipment id: IMEI_removed
>   ----------------------------------
>   System   |                 device: qcom-soc
>            |                drivers: rpmsg_ctrl, qcom-q6v5-mss, bam-dmux
>            |                 plugin: qcom-soc
>            |           primary port: wwan0qmi0
>            |                  ports: rpmsg_ctrl2 (ignored), wwan0 (net), wwan0at0 (at),  
>            |                         wwan0at1 (at), wwan0qmi0 (qmi), wwan1 (net), wwan2 (net), wwan3 (net),  
>            |                         wwan4 (net), wwan5 (net), wwan6 (net), wwan7 (net)
>   ----------------------------------
>   Status   |                   lock: sim-pin2
>            |         unlock retries: sim-pin (3), sim-puk (3), sim-pin2 (3), sim-puk2 (3)
>            |                  state: disabled
>            |            power state: off
>   ----------------------------------
>   Modes    |              supported: allowed: 3g; preferred: none
>            |                         allowed: 4g; preferred: none
>            |                         allowed: 3g, 4g; preferred: 4g
>            |                         allowed: 3g, 4g; preferred: 3g
>            |                current: allowed: 3g, 4g; preferred: 3g
>   ----------------------------------
>   Bands    |              supported: utran-1, utran-8, eutran-1, eutran-3, eutran-5, eutran-7,  
>            |                         eutran-8, eutran-20, eutran-38, eutran-40, eutran-41
>            |                current: utran-1, utran-8, eutran-1, eutran-3, eutran-5, eutran-7,  
>            |                         eutran-8, eutran-20, eutran-38, eutran-40, eutran-41
>   ----------------------------------
>   IP       |              supported: ipv4, ipv6, ipv4v6
>   ----------------------------------
>   3GPP     |                   imei: removed_from-here
>            |          enabled locks: fixed-dialing
>            |   packet service state: detached
>   ----------------------------------
>   3GPP EPS |   ue mode of operation: csps-1
>            |     initial bearer apn: internet.digimobil.es
>            | initial bearer ip type: ipv4
>   ----------------------------------
>   SIM      |       primary sim path: /org/freedesktop/ModemManager1/SIM/0
>            |         sim slot paths: slot 1: /org/freedesktop/ModemManager1/SIM/0 (active)
>            |                         slot 2: none
> ```

Just to verify did copy the modem firmware files in this way?

```bash
   *  mount -o loop -t vfat {modem_partition} /mnt
   * cd /mnt/image/
   * cp modem.*, wcnss.* mba.mbn /lib/firmware/  
   * In modem_pr/mcfg/configs/mcfg_sw/generic/, look for the appropriate mcfg_sw.mbn for your region (and carrier if applicable). Copy it to /lib/firmware.
```

alexandrglm, 2024-10-25:
Oh, nope. 
I didn't marked 'vfat' fs to mount, and also the apropiate mcfg_sw.mbm file.

I'll make a new attempt very soon; but I think it will work with your guidelines.

alexandrglm, 2024-10-26:
> Just to verify did copy the modem firmware files in this way?
> 
> ```shell
>    *  mount -o loop -t vfat {modem_partition} /mnt
>    * cd /mnt/image/
>    * cp modem.*, wcnss.* mba.mbn /lib/firmware/  
>    * In modem_pr/mcfg/configs/mcfg_sw/generic/, look for the appropriate mcfg_sw.mbn for your region (and carrier if applicable). Copy it to /lib/firmware.
> ```

It works! (But only when copying the entire partition files, idk why).

alexandrglm, 2024-10-26:
OpenWRT still doesn't want to work (to be flashed):

`/home/git# edl wf wrtWORKABLE.bin 
Qualcomm Sahara / Firehose Client V3.62 (c) B.Kerler 2018-2024.
main - Trying with no loader given ...
main - Waiting for the device
main - Device detected :)
main - Mode detected: firehose
modules
modules - [LIB]: 'Logger' object has no attribute 'loglevel'
Done |----------|   0.0% Write (Sector 0x0 of 0x738000) 0.00 MB/s                                                                                         Wrote wrtWORKABLE.bin to sector 0.
`

Do you like to talk about compiling OpenWRT for this device?

AlienWolfX, 2024-10-27:
> OpenWRT still doesn't want to work (to be flashed):
>
> `/home/git# edl wf wrtWORKABLE.bin Qualcomm Sahara / Firehose Client V3.62 (c) B.Kerler 2018-2024. main - Trying with no loader given ... main - [...]

Hii can you give this a try [OpenWRT](https://filebin.net/l98gz8zlgv0adhi6)

Steps:

`bash flash.sh`

once completed access `192.168.1.1` and flash `handsomemod-msm89xx-msm8916-Handsome_handsome-openstick-uz801-squashfs-sysupgrade.bin`

danyaPostfactum, 2024-10-27:
Hello @AlienWolfX. Thanks for your guide. I have installed openwrt-UZ801_v3.2.7z you provided. It works. But it seems LTE band is wrong because speed test shows ↓3.75/↑20.40Mbit (download speed is low)  . Can you point me how to view current LTE Band and manage cellular settings? I am new to OpenWRT. All i found is `mmcli -m 0`:
```
  Status   |                    lock: sim-pin2
           |          unlock retries: sim-pin (3), sim-puk (10), sim-pin2 (3), sim-puk2 (10)
           |                   state: connected
           |             power state: on
           |             access tech: lte
           |          signal quality: 100% (cached)
  -----------------------------------
  Modes    |               supported: allowed: 3g; preferred: none
           |                          allowed: 4g; preferred: none
           |                          allowed: 3g, 4g; preferred: 4g
           |                          allowed: 3g, 4g; preferred: 3g
           |                 current: allowed: 3g, 4g; preferred: 3g
  -----------------------------------
  Bands    |               supported: utran-1, utran-8, eutran-1, eutran-3, eutran-5, eutran-7,
           |                          eutran-8, eutran-20, eutran-38, eutran-40, eutran-41
           |                 current: utran-1, utran-8, eutran-1, eutran-3, eutran-5, eutran-7,
           |                          eutran-8, eutran-20, eutran-38, eutran-40, eutran-41
```

Also, i can connect to wifi access point only. I see the devices with name `RNDIS` and id `USB\VID_18D1&PID_D001&REV_0001&MI_00` requiring driver to work. Is it really RNDIS and what driver should i use?

alexandrglm, 2024-10-27:
Hi, @danyaPostfactum:

`Also, i can connect to wifi access point only. I see the devices with name RNDIS and id USB\VID_18D1&PID_D001&REV_0001&MI_00 requiring driver to work. Is it really RNDIS and what driver should i use?
`
RNDIS means the USB network connection you can enable with (not the LTE conn.); it's managed by kernel mod, try:

`
opkg install kmod-usb-net-rndis
reboot
'

For LTE things, wait for @AlienWolfX  response, but afai-understand, the band things belongs to "modem partition files already used in /lib/firmware".

Cheers.
Alexandr

AlienWolfX, 2024-10-27:
> Hello @AlienWolfX. Thanks for your guide. I have installed openwrt-UZ801_v3.2.7z you provided. It works. But it seems LTE band is wrong because speed test shows ↓3.75/↑20.40Mbit (download speed is [...]

Greetings, @danyaPostfactum
 
You need to install this [driver](https://github.com/milkv-duo/duo-files/raw/main/common/RNDIS_drivers_20231018.zip) in order for RNDIS to work on windows.

Regarding changing bands you can run `mmcli -m 0 --set-current-bands={BAND}`

alexandrglm, 2024-10-27:
> > OpenWRT still doesn't want to work (to be flashed):
> > `/home/git# edl wf wrtWORKABLE.bin Qualcomm Sahara / Firehose Client V3.62 (c) B.Kerler 2018-2024. main - Trying with no loader given ... main - Waiting for the device main - Device detected :) main - Mode detected: firehose modules modules - [LIB]: 'Logger' object has no attribute 'loglevel' Done |----------| 0.0% Write (Sector 0x0 of 0x738000) 0.00 MB/s Wrote wrtWORKABLE.bin to sector 0. `
> > Do you like to talk about compiling OpenWRT for this device?
> 
> Hii can you give this a try [OpenWRT](https://filebin.net/l98gz8zlgv0adhi6)
> 
> Steps:
> 
> `bash flash.sh`
> 
> once completed access `192.168.1.1` and flash `handsomemod-msm89xx-msm8916-Handsome_handsome-openstick-uz801-squashfs-sysupgrade.bin`
> 
> You will need to move the modem partition again to /lib/firmware to be able to use the modem also the WNCSS bin file for Wi-Fi
> 
> Building OpenWRT
> 
> Yes I would like to talk about compiling OpenWRT for this dongle as far as I've known it's based of [Handsomemod](https://github.com/HandsomeMod/HandsomeMod)

Well, my dongle has no storage consistance, it does not preserve any data after rebooting ... so, I'm stopping at this point, and reverting it to Debian openStick.
I'll resume the next week from this point.

You're a genius, sir.

Alexandr.

alexandrglm, 2024-10-27:
> Hii can you give this a try [OpenWRT](https://filebin.net/l98gz8zlgv0adhi6)
>
> Steps:
>
> `bash flash.sh`
>
> once completed access `192.168.1.1` and flash [...]

@AlienWolfX Firmly I confirm that my board has storage issues with this firm. So so so buggy, ha ha.
I'm reverting it to debian, I'll resume openwrt thins the next week.

You're a genius sir.

Alexandr.

alexandrglm, 2024-10-27:
@AlienWolfX Can you confirm the pads for unbricking the device?

https://github.com/AlienWolfX/UZ801-USB_MODEM/raw/main/img/Uz801_board.jpg

Just 2 wires on that pins? (It uses to be the regular way to boot EDL modes on qualcomm devices) but ... my device isn't want to load from the bootloader, and it's always starting to OpenWRT.

I'm also trying to short directly GND to usb d+ wire, no results.

alexandrglm, 2024-10-27:
> Hii can you give this a try [OpenWRT](https://filebin.net/l98gz8zlgv0adhi6)
>
> Steps:
>
> `bash flash.sh`
>
> once completed access `192.168.1.1` and flash [...]

**Device bricked. Don't use this.**
_Dongle trapped at non-functional openwrt thing, with no chances to get real EDL again._

AlienWolfX, 2024-10-27:
> @AlienWolfX Can you confirm the pads for unbricking the device?
> 
> https://github.com/AlienWolfX/UZ801-USB_MODEM/raw/main/img/Uz801_board.jpg
> 
> Just 2 wires on that pins? (It uses to be the regular way to boot EDL modes on qualcomm devices) but ... my device isn't want to load from the bootloader, and it's always starting to OpenWRT.
> 
> I'm also trying to short directly GND to usb d+ wire, no results. I've tested those pads, neither both has GND continuity.
> 
> Any idea?

Yes If doesn't work you can run the commands below to get edl working:

```bash
adb reboot-bootloader

fastboot oem reboot-edl
```

AlienWolfX, 2024-10-27:
> @AlienWolfX Can you confirm the pads for unbricking the device?
>
> https://github.com/AlienWolfX/UZ801-USB_MODEM/raw/main/img/Uz801_board.jpg
>
> Just 2 wires on that pins? (It uses to be the regular way [...]

That's odd did you short the pads or the D+ to GND before plugging in the dongle.?

alexandrglm, 2024-10-27:
> > @AlienWolfX Can you confirm the pads for unbricking the device?
> > https://github.com/AlienWolfX/UZ801-USB_MODEM/raw/main/img/Uz801_board.jpg
> > Just 2 wires on that pins? (It uses to be the regular way to boot EDL modes on qualcomm devices) but ... my device isn't want to load from the bootloader, and it's always starting to OpenWRT.
> > I'm also trying to short directly GND to usb d+ wire, no results. I've tested those pads, neither both has GND continuity.
> > Any idea?
> 
> Yes If doesn't work you can run the commands below to get edl working:
> 
> ```shell
> adb reboot-bootloader
> 
> fastboot oem reboot-edl
> ```
> 
> That's odd did you short the pads or the D+ to GND before plugging in the dongle?.

Solved the adb - fastboot - edl issue. I thought ADB wasn't available at openwrt img sys.

So strange the pad issue, I use to edl qualcomm phone shorting pads gnd - d+.

I've shorted here:

1. Those two pads (the nearest 2 pads from the chasis, marked as EDL at the picture given).
2. The chasis to both pins separately, and also the 2 joind with the pad.
3. The USB GND to D+, and also to D-.
None of these worked.

Also, I've cheked the elec. continuity from GND to EACH pad on the board with a tester, none of them has GND continuity.

I don't know if it's my board faulty, or a "newer baord revision" thing.

danyaPostfactum, 2024-10-27:
> You need to install this [driver](https://github.com/milkv-duo/duo-files/raw/main/common/RNDIS_drivers_20231018.zip) in order for RNDIS to work on windows
Ok, it looks like preinstalled Microsoft USB RNDIS driver works fine (I choosed it manually)

@alexandrglm I've just shortened these 2 pads with tweezers and then inserted modem into usb. It forces EDL mode.
But "D+" trick does not work.

alexandrglm, 2024-10-27:
> @alexandrglm I've just shortened these 2 pads with tweezers and then inserted modem into usb. It forces EDL mode. But "D+" trick does not work.

Mine's not working when shorting pads.
PAD1 offers 320Ohm resistance against GND.
So, I shouldn't kid with the aboot,sbl and boot.

danyaPostfactum, 2024-10-27:
> I'm reverting it to debian

Do you use wifi hotspot? Is speed ok? OpenWRT limits wifi tx speed to 6.5Mbit/s by some reason...

alexandrglm, 2024-10-27:
> > I'm reverting it to debian
> 
> Do you use wifi hotspot? Is speed ok? OpenWRT limits wifi tx speed to 6.5Mbit/s by some reason...

I hadn't any issue LTE band related using Debian, the unique limit the coverage speed given by the ISP.

danyaPostfactum, 2024-10-28:
> It works! (But only when copying the entire partition files, idk why).

Hi. Could you explain how to do this? What is {modem_partition}? Should i run this on device ssh? Or should i prepare rootfs.img before flashing? I don't understand. Sorry, i have no linux experience.

I'm trying to use Debian, but lte connection doesn't work. I don't understand how to copy modem files to /lib/firmware/ 

> 70Mbps but after flashing well my modem partition files

I flashed it already (otherwise lte doesn't work at all).

> PAD1 offers 320Ohm resistance against GND.

It seems to be burned. Correct value should be at least 37K Ohm for this pad and 216K for the second pad.

alexandrglm, 2024-10-28:
> Hi. Could you explain how to do this? What is {modem_partition}? Should i
run this on device ssh? Or should i prepare rootfs.img before flashing? I
don't understand. Sorry, i have no linux experience.

Modem partition, in Android devices, are 3 partitions which stores all the
LTE/UMTS/GSM configs, WLAN configs and BT config, also the IMEI, BT/WLAN
MAC address and other serial numbers of your device.

This device is like an Android phone, so, you have inside the modem.img
partition (what we need to extract, here), and the other 2 modemst
partitions (which we don't need to do nothing).

First of all, you should have made a full backup of the nand flash using
Bkerler's EDL soft (or another soft which can work with Qualcomm devices in
EDL mode). Did you made that backup? In that backup you can find the
mentioned "modem partition". And, yes, you can extract its files. Those
files are what you need to copy at /lib/firmware/. At that point, the
dongle can work with the LTE.

It's recommendable using linux for all of this stuff, but it also works on
windows. Apart from this, win has linux terminal inside so, you'll
definitelly able to get this.

Tell us if you already have backuped your entire flash, to continue the
explaination and howto's. :)

> It seems to be burned. Correct value should be at least 37K Ohm for this
pad and 216K for the second pad.
Lol ! It may be burnt, the qualcomm cpu has 4 cores. Previous firmware
version disable two cores to avoid, literally, buring the device....but my
firmware is different from mentioned here so, it may be burnt, but it works!

I

alexandrglm, 2024-10-28:
> @alexandrglm I have my backup. I followed insturctions )). Thanks for quick explanation about modem partitions. The problem is that I don't understand how to copy local files into the running device. I has no adb for push/pull ;)

If you have SSH, you should give a try to SCP (SSH pull/push method):
https://linuxize.com/post/how-to-use-scp-command-to-securely-transfer-files/

At least on linux I used:

**Pushing files to device:**

# scp {user}@ip_host:{path_of_file} {destination_folder_on_the_device}

E.g.

```
# scp /home/user/modem.b01 root@192.168.1.1:/lib/firmware/
# scp /home/user/modem.b02 root@192.168.1.1:/lib/firmware/
```
.... and one by one, all the files needed.

Windows should be something like....
scp "C:\path\of\file\modem.b01" user@ip_host:/lib/firmware/

The same, but the PATHS from WIN pc needs to be like this "\" (not "/")
Give a try using a GUI apps like WinSCP or Putty.

On the other hand, Modem partition (its files) has to be extracted before. 
Have you already extracted your modem partition from the entire 3,6Gb block backup bin file? 
If not, it has to be mounted and extracted. I don't know how to do this at Windows without using any external software, but you may:

1. Use the linux terminal on windows, and the copy the files to win in order to SCP them (or SCP them inside the linux chroot terminal of Windows).
2. Maybe, an external software can help you, but it's really easy using one command line at linux, and maybe at windows. Why? Because the modem partition has FAT format, so, it's compatible with win. Maybe a "mounting images software".

Command to mount, as @AlienWolfX helped sharing it:

*  mount -o loop -t vfat {modem_partition} /mnt
   * cd /mnt/image/
   * cp modem.*, wcnss.* mba.mbn /lib/firmware/  
   * In modem_pr/mcfg/configs/mcfg_sw/generic/, look for the appropriate mcfg_sw.mbn for your region (and carrier if applicable). Copy it to /lib/firmware.

alexandrglm, 2024-10-28:
@danyaPostfactum If SCP is not working, you can zip the files and upload it to a host which gives you a direct link to download the zip.

Then, in device, connected to inet, wget the url link, then unzip to a folder, and done.

danyaPostfactum, 2024-10-28:
@alexandrglm I've extracted modem files using 7zip.

> Then, in device, connected to inet

How to connect? ;) The reason I am trying to copy the files is a broken cellular connection.

> Modem partition (its files) has to be extracted before

I just use 7zip to extract.

[image]

Password for `user` is `1`. But it has no permission. What password for `root`? I have no idea...

danyaPostfactum, 2024-10-28:
@alexandrglm thank you for your help. I've just realised that i can use modem as wifi client to get internet access. I will deal with this now ))

alexandrglm, 2024-10-28:
> @alexandrglm I've extracted modem files using 7zip.
> 
> > Then, in device, connected to inet
> 
> How to connect? ;) The reason I am trying to copy the files is a broken cellular connection.
> 
> > Modem partition (its files) has to be extracted before
> 
> I don't know if dongle's debian already comes with p7zip-full binaries. If not, tar will also help you.
> 
> ![image](https://private-user-images.githubusercontent.com/1527097/380848976-f2fec943-aa52-44f2-a1e3-1b5c519b9027.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzAxNDE5NjYsIm5iZiI6MTczMDE0MTY2NiwicGF0aCI6Ii8xNTI3MDk3LzM4MDg0ODk3Ni1mMmZlYzk0My1hYTUyLTQ0ZjItYTFlMy0xYjVjNTE5YjkwMjcucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTAyOCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDEwMjhUMTg1NDI2WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9YjMwYTk5OGJhZGY4NDkyOTk1N2Y1YWI4ZjYyYTMwMzA3NGNkZDJjMTZiMjFlNzQ1NmNmMmExZTYwZTI2YzI5NiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.wx6UVARLnmY-Ra_a3g1Irh4C7NFvvMWcBTlmwLzDFdo)
> 
> Password for `user` is `1`. But it has no permission. What password for `root`? I have no idea...

Connect with ssh as user.
Then make a sudo instante and change the root password:
`$ sudo su
Pasword for user?: Type 1
# passwd root
Change the root password.`

Root account comes disabled, but you can active it and configure your pwd like this.
Then, make a SCP method.

If Windows is showing the 'Permission denied' message again, open a Terminal with admin rights.

danyaPostfactum, 2024-10-28:
> Root account may come disabled, but you can active it and configure your pwd like this.

I have set up new password for root. But I can't connect to root anyway...
[image]

Thanks anyway, I will try to use wifi to download files..

alexandrglm, 2024-10-28:
> I have set up new password for root. But I can't connect to root anyway... ![
> Thanks anyway, I will try to use wifi to download files..

Strange. I don't know exactly why but I used mediafire (and copied the direct download link, not the regular sharing link) to get the .zip at the device. It's OK using wget but if I hadn't any inet connection, how could I fix it?
Sure SCP binary inside openstick debian works, if not, it could be an users-groups/rights issue to fix.

I'm not using debian at this point because I'm testing some checks at Android.

Cheers.

danyaPostfactum, 2024-10-28:
> I'm testing some checks at Android

Are you trying to disable wifi on stock firmware?

My primary issue was a wifi connection problem (extremely slow connection with my android tv box). I tried to change wifi channel (by modifying /data/mificonfig.json), it did not help. It looks like I am luck to have buggy tv box wifi + buggy modem wifi.
With openWRT this problem disappears (but openWRT adds another issues).

alexandrglm, 2024-10-28:
Do you want my "truth"?
I LOVE OpenWRT, I'm using it on many devices, but I'm also getting errors
'cause this device isn't fully compatible (yet)
Debian is fine for me, it works well on my board. But ...

The stock Android firmware works perfect. At least at a "Facilities of use"
point of view. No Wifi problems. No LTE problems.
The webgui is so crapped. But it works (And, no, I can't switch off the
Wifi via webgui, but it can be disabled if needed using nmcli commands, or
editing a file via ssh).
It came already rooted. I'm working on testing if the stock firm comes
"clean", or it's a risky firmware.

If your tvbox has an USB port and it's able to work with dongles via RNDIS,
and your goal is setting up easily, and not debugging the dongle ..... if
your stock firm fits your needings ... Use the stock firm. Disable the wlan
via shell at ssh. And that's it.

AlienWolfX, 2024-10-28:
> Password for `user` is `1`. But it has no permission. What password for `root`? I have no idea...
> 

Hello,

Password for root is `1`

danyaPostfactum, 2024-10-29:
New day and new issues...
@alexandrglm  hi, i've uploaded modem files (I creaded local webserver using nodejs). LTE works. But wifi hotspot is buggy.
Devices reconnect every minute. ATV box doesn't work at  all (it worked for the first time, but with 1mbit issue again).
But connection with my laptop works good. It's weird.. I think I will try openwrt again. Or maybe could you share your debian image with me (if you dumped it)? I will replace IMEI.

> I didn't notice the "band" problem

Maybe I was wrong. Most of time it works good.

> Password for root is `1`

Hi. I can connect to user@192.168.200.1 with password 1, but no luck with root@192.168.200.1

alexandrglm, 2024-10-29:
> Hi. I can connect to ***@***.*** with password 1, but no luck with

No possible at all this week, I can't focus so much time with UZ801; It's
still flashed on Android.
About IMEI, well, don't worry about, it's so ridiculous the way imei are
configured. A dump could be sent with blank serials. Tell @AlienWolfX if it
has an openstick dump available.

About yout tv-box and the dongle; 1Mbps wlan link it's like no 802.11n
protocol involved, but the dongle is capable to run at n speed. If you
think about what modem-wlan configs you are using (its own config), the
problem seems to be at the tv-box side.

Is it using openwrt, or linux based os? Look at /etc/config/network file ->
option interfaces/devices ... forcing the wlan.radio (or what you named it)
the 802.11 protocol as seen in
https://openwrt.org/docs/guide-user/network/wifi/basic

I can't explain why but sometimes devices/connectivity, at "defaults"
configs, tend to select the more stable way, the better (E.g., the LTE
issue, uploading bandwith higher than dwn, and, suddenly, download bandwith
is the higher; etc.).

danyaPostfactum, 2024-10-29:
> Is it using openwrt, or linux based os?

Android tv. It works good with any other wifi hotspots. It works bad (less then 1mbit/s download but quite qood upload speed) with stock UZ801 firmware and apparently debian os (not sure because debian hotspot does not work correctly). It works good with UZ801 openWRT (but openWRT has issue with 6Mbit host-to-client speed limit: `6.5 Mbit/s, 20 MHz, MCS 0`, somehow i can get `13.0 Mbit/s, 20 MHz, MCS 1` or `13.5 Mbit/s, 40 MHz, MCS 0,` playing with hotspot config, but it works until hotspot reset)

>- If android/linux based, try some shell commands:

thanks i will try. But i guess this issue shoudl be fixed on the wifi hotspot side (openWRT showed the best result)

danyaPostfactum, 2024-10-29:
>  iw dev {your_wlan_interface} set type n

I have 0,8Mbit download/12Mbit upload speed with the stock UZ801. So, i think the mode is already `n`

alexandrglm, 2024-10-29:
> thanks i will try. But i guess this issue shoudl be fixed on the wifi hotspot side (openWRT showed the best result)

So, try to force the wlan AP interface at N rates on debian-dongle.
I can't check it at this moment, but, a general way may be:

1. Editing the hostapd conf file /etc/hostapd/hostapd.conf  (try to find it using f_ind /  -name "conf"| grep 'hostapd'_  or with  _-name "*hostapd*_"  :

_interface={the_interface_given_name}

driver=nl80211 /* or what kernel has chosen for the wlan driver */

ssid= {your network ap name}

hw_mode=g       /* This enables 802.11 mixed mode, the "gate to n rates when available" */

channel={the selected ch}

ieee80211n=1          /* This enable 802.11n */

ht_capab=[HT40+][SHORT-GI-20][SHORT-GI-40]  /* some specific configs ways to choose the 802.11 propagation method linked to the 802.11 g/n rates */

wpa=2 /* wpa3 psk encryption, or what encryption available method you selected */

wpa_passphrase={your_network_passwd}_

2. Using iw binary as mentioned before (_iw dev {wlanAP_interface_name} set type n_ on root account or doing sudo).

alexandrglm, 2024-10-29:
> > iw dev {your_wlan_interface} set type n
> 
> I have 0,8Mbit download/12Mbit upload speed with the stock UZ801. So, i think the mode is already `n` (because of good upload speed)

Higher dwn rates than up rates on **wlan** device?
My head's exploding. Idk where to find traces at this moment.

AlienWolfX, 2024-10-29:
> Hi. I can connect to user@192.168.200.1 with password 1, but no luck with root@192.168.200.1
> 

Hi,

Direct ssh via `root@192.168.200.1` doesn't work with openstick, Thus I suggest you to use sudo or su with `1` as password if you intend to use the root user.

Best regards,

Allen

[58 more replies]

closed 2025-10-26 by alexandrglm