AlienWolfX/UZ801-USB-MODEM #9: IMEI lost after Flashing UZ801

Irshadsamejo, 2024-11-04:
Hi everyone after flashing stock firmware on UZ801 usb stick version 3.2 i am unavailable to catch 4g signals in web interface IMEI showing 0.

Is there any way to fix this

alexandrglm, 2024-11-04:
> Hi everyone after flashing stock firmware on UZ801 usb stick version 3.2 i am unavailable to catch 4g signals in web interface IMEI showing 0.
> 
> Is there any way to fix this

Yes. If the web GUI does not allow you to change nothing, you should restore your modem1st&modemst2 partitions previously backuped!

If there are no backups try QPST Tools (Windows) in order to restore your serials ... QCN files can be edited directly with a hex editor, or using some easy tools for the matter.

Look at this guide, the steps are similar to all Qualcoms.
[XDA - Fix lost serials ...](https://xdaforums.com/t/guide-backup-edit-and-restore-qcn-fixing-lost-imei.4101611/)

But firstly try with the webGUI, and **remember not to forget doing full backups before flashing anything.**

Irshadsamejo, 2024-11-04:
> > Hi everyone after flashing stock firmware on UZ801 usb stick version 3.2 i am unavailable to catch 4g signals in web interface IMEI showing 0.
> > 
> > Is there any way to fix this
> 
> Yes. If the web GUI does not allow you to **restore**  _*_ nothing, you should restore your modem1st&modemst2 partitions previously backuped!
> 
> If there are no backups try QPST Tools (Windows) in order to restore your serials ... QCN files can be edited directly with a hex editor, or using some easy tools for the matter.
> 
> Look at this guide, the steps are similar to all Qualcoms.
> [XDA - Fix lost serials ...](https://xdaforums.com/t/guide-backup-edit-and-restore-qcn-fixing-lost-imei.4101611/)
> 
> But firstly try with the webGUI, and **remember not to forget doing full backups before flashing anything.**
> 
> 
> _*_ Qualcomm chipsets usually are only allowed to be RESTORED  (not changed) to its original state. Otherwise, the modem is not available. Cheers.
> 
Hi bro first of all thank you for your reply 

I purchased this stick a month ago and it was working fine with 4g signals with good speed for some reasons I wanted to flash OpenWRT so backed up my firmware with the help of Miko tool and Then flashed OpenWRT the file that I flashed was a buggy one and hence my device didn't rebooted then I flashed my stock firmware( **not a .bin file there was a list of partitions**)back but after flashing this time device struck on red light then I tried different firmwares including the file that you uploaded and another firmware file from @AlienWolfX in both firmwares my device rebooted successfully but shows Server provider's name but not signals and IMEI showing 0

alexandrglm, 2024-11-04:
> Hi bro first of all thank you for your reply
>
> I purchased this stick a month ago and it was working fine with 4g signals with good speed for some reasons I wanted to flash OpenWRT so backed up my [...]

If you really have a backup of your original firmware, it's much better if it's partition by partition! 
You just have to flash, either by EDL, by fastboot, or by whatever tool you used to make the backup, 
the "modem", "modemst1" and "modemst2" partitions.

If this doesn't work for you at all, restore the "fsc" and "fsg" partitions as well.

When working with foreign dumps, you should always do a "frankenstein nand" combining most of the partitions from that foreign dump with your minimum own partitions (modem, modemst1, modemst2 ... fsc, fsg) so that your system doesn't lose its serials.

Hope this helps.
Alexandr.

P..S. What model do you have? What firmware version - what date is it?
If you feel like using the dumpsys_tool to report your firmware here, remember to remove sensitive data (mac's, ip's, IMEI's, serial, hostname) before sharing. Regards.

alexandrglm, 2024-11-04:
> Hi bro first of all thank you for your reply
>
> I purchased this stick a month ago and it was working fine with 4g signals with good speed for some reasons I wanted to flash OpenWRT so backed up my [...]

This script can be used with the Bjoern Kerler's EDL tool in order to ERASE and RESTORE your modem's/fsc/fsg.
Configure your locals.

```
#!/bin/bash

if ! command -v edl_cmd &> /dev/null; then
    echo "ERROR: Check EDL, chroot, python env, etc..."
    exit 1
fi

output_dir="./output" 
edl_cmd="/usr/local/bin/edl" # CONFIRURE YOUR LOCALS EDL BINARY PATH ....  $ which edl 

declare -A partitions=(
    ["modem"]="modem.img" # PUT THE REAL MODEM IMG NAME
    ["modemst1"]="modemst1.img" # PUT THE REAL MODEMST1 IMG NAME
    ["modemst2"]="modemst2.img" # PUT THE REAL MODEMST2 IMG NAME
    ["fsc"]="fsc.img"
    ["fsg"]="fsg.img"
)

for partition in "${!partitions[@]}"; do
    img_file="$output_dir/${partitions[$partition]}"
    if [[ ! -f "$img_file" ]]; then
        echo "ERROR: There's no $img_file to be flashed at $partition."
        exit 1
    fi
done

for partition in "${!partitions[@]}"; do
    img_file="$output_dir/${partitions[$partition]}"
    echo "Erasing $partition ..."
    edl e "$partition"
    
    echo "Writing $img_file at $partition ..."
    edl w "$partition" "$img_file"
done

echo "Done!"
```

The same, but using fastboot, copy these partitions and change its extension to .img :

```
#!/bin/sh

echo "Entering fastboot mode...."
adb reboot bootloader
sleep 5

for x in modem modemst1 modemst2 fsc fsg; do # here you can add
    fastboot erase ${x} ${x}.img
for x in modem modemst1 modemst2 fsc fsg; do # here you can add
    fastboot flash ${x} ${x}.img
done
```

Irshadsamejo, 2024-11-05:
> > > > Hi everyone after flashing stock firmware on UZ801 usb stick version 3.2 i am unavailable to catch 4g signals in web interface IMEI showing 0.
> > > > Is there any way to fix this
> > > 
> > > 
> > > Yes. If the web GUI does not allow you to **restore**  _*_ nothing, you should restore your modem1st&modemst2 partitions previously backuped!
> > > If there are no backups try QPST Tools (Windows) in order to restore your serials ... QCN files can be edited directly with a hex editor, or using some easy tools for the matter.
> > > Look at this guide, the steps are similar to all Qualcoms.
> > > [XDA - Fix lost serials ...](https://xdaforums.com/t/guide-backup-edit-and-restore-qcn-fixing-lost-imei.4101611/)
> > > But firstly try with the webGUI, and **remember not to forget doing full backups before flashing anything.**
> > > _*_ Qualcomm chipsets usually are only allowed to be RESTORED  (not changed) to its original state. Otherwise, the modem is not available. Cheers.
> > 
> > Hi bro first of all thank you for your reply
> > 
> > I purchased this stick a month ago and it was working fine with 4g signals with good speed for some reasons I wanted to flash OpenWRT so backed up my firmware with the help of Miko tool and Then flashed OpenWRT the file that I flashed was a buggy one and hence my device didn't rebooted then I flashed my stock firmware( **not a .bin file there was a list of partitions**)back but after flashing this time device struck on red light then I tried different firmwares including the file that you uploaded and another firmware file from @AlienWolfX in both firmwares my device rebooted successfully but shows Server provider's name but not signals and IMEI showing 0
> 
> If you really have a backup of your original firmware, it's much better if it's partition by partition! 
> You just have to flash, either by EDL, by fastboot, or by whatever tool you used to make the backup, 
> the "modem", "modemst1" and "modemst2" partitions.
> 
> If this doesn't work for you at all, restore the "fsc" and "fsg" partitions as well.
> 
> When working with foreign dumps, you should always do a "frankenstein nand" combining most of the partitions from that foreign dump with your minimum own partitions (modem, modemst1, modemst2 ... fsc, fsg) so that your device doesn't lose its serials.
> 
> Hope this helps.
> Alexandr.
> 
> P..S. Which model do you have? What firmware version - what date is it?
> If bought one month ago, your firmware might be the same I have (2024 Apr, 26 version) or newer!
> If you feel like using the dumpsys_tool to report your firmware here, remember to remove sensitive data (mac's, ip's, IMEI's, serial, hostname) before sharing. Regards.
> 

Hi bro how to back up firmware in .bin file

alexandrglm, 2024-11-05:
> Hi bro how to back up firmware in .bin file

Be more specific, please:

Backuping using what tool?
EDL? Fastboot?

Backuping the entirne nand flash on a big block .bin, or each partition on .bin files?

alexandrglm, 2024-11-06:
@Irshadsamejo I've heard about Miko tool before, but I've never used it (I don't know if it's a Win-only app).
The Bjoern Kerler's EDL tool is my favourite when using Qualcomm' devices.

Those scripts cna be used as an automotive way to make partitions, but the simpler, the better.

EDL commands to make **full NAND backup**:   

```
$ edl rf name_of_file.bin  --gen-xml  //.bin, .img, it does not care, you'll get a huge block file with entire nand.  "rf" syntax is so intuituve ... read-flash. --gen-xml will make a rawprogram0.xml file, backup it too
```

EDL commands to backup **partition by partition**:  
```
$ edl rl --gen-xml  // 
```

Partitions can also be read by using fastboot, but I've tested if both ways gives the same backups byte by byte.
I think EDL is the best way.

This is the[ guide](https://github.com/bkerler/edl/blob/cef0076e1d4d29c1887c51786eb588503657b907/README.md).

Alexandr.

Irshadsamejo, 2024-11-07:
> @Irshadsamejo I've heard about Miko tool before, but I've never used it (I don't know if it's a Win-only app).
> The [Bjoern Kerler's EDL tool ](https://github.com/bkerler/edl) is my favourite when using Qualcomm' devices.
> 
> Those scripts can be used as an fast way to do some stuff directly to device, but the simpler, the better.
> 
> EDL commands to make **full NAND backup**:   
> 
> ```
> $ edl rf name_of_file.bin  --gen-xml       //.bin, .img, it does not care, you'll get a huge block file with entire nand.  "rf" syntax is so intuituve ... read-flash. --gen-xml will make a rawprogram0.xml file, back it up too.
> ```
> 
> EDL commands to backup **partition by partition**:  
> ```
> $ edl rl --gen-xml 
> ```
> 
> Partitions can also be read by using fastboot, but I haven't tested yet if both methods return identical partitions byte by byte.  
> 
> Again, it doesn't matter which extension the backups have, this doesn't modify a dump in itself. 
> What matters is that the dump is done correctly.  
> 
> Here is the[ guide](https://github.com/bkerler/edl/blob/cef0076e1d4d29c1887c51786eb588503657b907/README.md).
> 
> Alexandr.
> 
> 
> 
> 
> 
Thanks bro even though I have .bin file but still i am unable to access 4g network

AlienWolfX, 2024-11-07:
> Thanks bro even though I have .bin file but still i am unable to access 4g network 
> 

Hello @Irshadsamejo 

Did you manage to fix your lost IMEI?.

alexandrglm, 2024-11-07:
> Thanks bro even though I have .bin file but still i am unable to access 4g network

You should try editing a blank QCN file, but with the IMEI's provided for your device, through QPSTools
There are some QCN blank files for UZ801 dongle and HowTo's at some chinese forums.
It simpler that how it sounds.

Irshadsamejo, 2024-11-07:
> > Thanks bro even though I have .bin file but still i am unable to access 4g network 
> > 
> 
> Hello @Irshadsamejo 
> 
> Did you manage to fix your lost IMEI?.
> 
> Ps. ***I've accidentally edited your comment XD***
> 

No bro not yet looks like I am going to give up 😞

Irshadsamejo, 2024-11-07:
> > Thanks bro even though I have .bin file but still i am unable to access 4g network
> 
> You should try editing a blank QCN file, but with the IMEI's provided for your device, through QPSTools.
> There are some QCN blank files for UZ801 dongle and HowTo's [here.](https://www.123pan.com/s/3GL7Vv-ur4Sh.html) and [here](https://github.com/alexandrglm/UZ801_stock_minitools/blob/107a74ad1197c5034450c65717969c7de015a34a/dumpsys_analysis/UZ801.qcn).
> 
> 
> 
> It seems they use a typical CH341A programmer to connect to the dongle via serial (Tx, Rx, GND), but really:
> any Qualcomm (old) device like this can be put into DIAG+AT+MODEM mode over USB to make these modifications:
> 
> 1. **.Simple EDL mode should be enough.** 
> 2. ```$ adb shell setprop sys.usb.config diag,diag_mdm,diag_mdm2,qdss,qdss_mdm,serial_cdev,dpl,rmnet,adb``` 
> 3. Also, try this: http://192.168.100.1/ms.html
> 
> It sounds more difficult than it actually is.
> 
> 
> P.S.: What @AlienWolfX have asked you for is the key .....Is the IMEI issue still unresolved? Or, are you now unable to access 4G despite having your IMEI restored? 
> 

Ok thanks i will try this

alexandrglm, 2024-11-07:
> > > Thanks bro even though I have .bin file but still i am unable to access 4g network
> > 
> > 
> > Hello @Irshadsamejo
> > Did you manage to fix your lost IMEI?.
> > Ps. _**I've accidentally edited your comment XD**_
> 
> No bro not yet looks like I am going to give up 😞

No sir, don't give it up!
You have the tools and ways to do it, here, explained. 
If one method fails, use another. 
If a version of QPST doesn't work quite right, try another one (Older versions worked for me, the latest 2.7.x gave me an error). 
If a Qualcomm driver doesn't work, search a little more and find the ones that do work. 
Install the drivers in Win in Test Mode.

With all this, you get it.

Irshadsamejo, 2024-11-07:
> > > > Thanks bro even though I have .bin file but still i am unable to access 4g network
> > > 
> > > 
> > > Hello @Irshadsamejo
> > > Did you manage to fix your lost IMEI?.
> > > Ps. _**I've accidentally edited your comment XD**_
> > 
> > No bro not yet looks like I am going to give up 😞
> 
> No sir, don't give it up!
> You have the tools and ways to do it, here, explained. 
> If one method fails, use another. 
> If a version of QPST doesn't work quite right, try another one (Older versions worked for me, the latest 2.7.x gave me an error). 
> If a Qualcomm driver doesn't work, search a little more and find the ones that do work. 
> Install the drivers in Win in Test Mode.
> 
> With all this, you get it.
> 
> But if you gonna finally throw it away, give it to me as a gift via airmail! (Lol, just kidding. Just make a fixing plan, and fix it).
> 
> Regards.
> 

Hahaha thanks bro busy a little bit due to my interview preparation once I will try to get Modem back if can't Then i will send to you although I am living in backward area services like airmail are not available here yet i will try my best 😄

Irshadsamejo, 2024-11-08:
> Ok thanks i will try this

Thanks dear i had made little bit progress thanks for your help earlier there was no option to restore backup qcn But after this method i was able to put device in DIAG mode but still there is a issue i have attached the log file [image]

AlienWolfX, 2024-11-08:
> Thanks dear i had made little bit progress thanks for your help earlier there was no option to restore backup qcn But after this method i was able to put device in DIAG mode but still there is a issue i have attached the log file

It appears that the error is caused by a  missing `QMSL_MSVC10R.dll` file have you tried using another up to date version of the application?.

Best regards,
Allen

Irshadsamejo, 2024-11-08:
> > Thanks dear i had made little bit progress thanks for your help earlier there was no option to restore backup qcn But after this method i was able to put device in DIAG mode but still there is a issue i have attached the log file
> 
> It appears that the error is caused by a  missing `QMSL_MSVC10R.dll` file, have u already tried using another up to date version of the application?.
> 
> Best regards,
> Allen
> 
Thanks Allen Which version should I try can you give me the link of that file

AlienWolfX, 2024-11-08:
> Thanks Allen Which version should I try can you give me the link of that file

I'll give an update regarding this once I start my experimentation.

alexandrglm, 2024-11-08:
- What's really happening here is that you're missing a library. Check the QPST installer, see if you've downloaded an untrusted version.
This is going to happen to you a lot, and you'll probably end up wanting to format your PC after installing all the software and fixing the dongle.

- MSVC10R.dll is a common dll used on Windows; your QPST is using its own MSVC10R.dll version, aparentlly in order to get working, but maybe your Win version is now incompatible with this; try to find MSVC10R.dll (with no QML at the beggining) on your /Windows folder and make a copy/move the copy to the QPST running tool which has the original .dll, and rename it with the orig. name.
- Install vscredist2010.exe / vscredistXversion.exe from Microsoft sources.

- Else, try another QPST version. It really happens, sometimes the latest version 2.7 doesn't work, and v2.1.x does. Sometimes a zip or exe from a website is corrupted and the same version doesn't....

- You was told by @AlienWolfX to test the modem_at commands, **it really works when restoring IMEI's.**. Hope this command serves you !

AlienWolfX, 2024-11-08:
> Thanks Allen Which version should I try can you give me the link of that file

Hello @Irshadsamejo,

`QPST-2.7.437.7z` works for me kindly follow this [guide](https://xdaforums.com/t/guide-backup-edit-and-restore-qcn-fixing-lost-imei.4101611/) mentioned by @alexandrglm it has the download link to all of the software's that you'll be using. Your task involves:

> [!IMPORTANT]
> Carefully do the task one by one and if you have questions feel free to ask

1. Backing up your QCN (make sure **Enable Multi-SIM** is selected)
2. Open your qcn backup with **IMEIRebuilder.exe** and add your IMEI 
3. Restore your QCN with QPST and reboot your device.

Irshadsamejo, 2024-11-08:
> Hello @Irshadsamejo,
>
> `QPST-2.7.437.7z` works for me kindly follow this [guide](https://xdaforums.com/t/guide-backup-edit-and-restore-qcn-fixing-lost-imei.4101611/) mentioned by @alexandrglm it has [...]

Thanks to both of you 
There is a little bit progress after installing QPST the earlier error is gone now when I successfully back up my .qcn file i tried to open with IMEI rebuilder tool it's given an error of unhandling but there was an other.qcn file with pre existing IMEI when I flashed this I got IMEI back but its a different one than device's original Imei

Irshadsamejo, 2024-11-08:
> > Thanks Allen Which version should I try can you give me the link of that file
> 
> Hello @Irshadsamejo,
> 
> **UPDATE**
> 
> It seems that `modem_at` is universally present across different versions of the firmware thus the solution to fixing lost IMEI can be solved with `modem_at AT+WRIMEI={YOUR_IMEI}` Hooray 🎆 
> 
> > [!IMPORTANT]
> Although setting IMEI works if you flashed a stock dump which is not covered with your region you'll encounter the `UNKNOWN` network problem, I think there is a way to solve this, I'll be updating this response once I got my answer.
> 
> Best regards,
> Allen
> 

You are right maybe issues is with firmware

AlienWolfX, 2024-11-08:
> You are right maybe issues is with firmware

You got `modem_at` working bro?

alexandrglm, 2024-11-08:
Are there available stock roms for this dongle?

AlienWolfX, 2024-11-09:
> Are there available stock or customs rom for this dongle?

Currently, there are no known stock or custom ROMs available for this dongle. Maybe a deep dive in baidu or some other sites can yield some result that may point to it's original SDK or some custom roms.

Irshadsamejo, 2024-11-13:
> > Are there available stock or customs rom for this dongle?
> 
> Currently, there are no known stock or custom ROMs available for this dongle. Maybe a deep dive in baidu or some other sites can yield some result that may point to it's original SDK or some custom roms.

Any update

AlienWolfX, 2024-11-13:
> Any update

Still looking for a way to unlock network

alexandrglm, 2024-11-13:
Unlocking the network? Means IMEI issue fixed but modem compromised?

alexandrglm, 2024-11-13:
> Yes, the IMEI is fixed, but the downside is that it doesn't recognize the
> network because he's using a firmware from a different region. I also
> experimented with this using my dongle. My guess is that it involves the
> mcfg_sw.mbn file or some other settings.

Sure, assuming you're trying OpenWRT or Debian, you'd choose the appropiate
mcfg_sw.mbn file, from the correct region folder, when extracting modem.img
data.

I'm glad you solved the imei fix.

Irshadsamejo, 2024-11-14:
Oh! Very sad
Can i still get signals with OpenWRT or debian

AlienWolfX, 2024-11-14:
> Oh! Very sad Can i still get signals with OpenWRT or debian

@Irshadsamejo Yes, you may use the OpenWRT build under release.

closed 2024-11-14 by AlienWolfX

Irshadsamejo, 2024-11-14:
> > Oh! Very sad Can i still get signals with OpenWRT or debian
> 
> @Irshadsamejo Yes, you may use the OpenWRT build under release as it is plug and play.

Where is that build

Irshadsamejo, 2024-11-14:
My Device (IMEI ) IMEI was not compliment with our telecommunication authority knowing this i have paid the tax for that as well but still can't get signals

AlienWolfX, 2024-11-14:
> Where is that build

@Irshadsamejo You can find it under releases, alternatively you can click [this](https://github.com/AlienWolfX/UZ801-USB_MODEM/releases/download/v1.0.0-alpha/openwrt-UZ801.tar.xz)
[image]