AlienWolfX/UZ801-USB-MODEM #10: A few observersions:

tboege, 2024-11-06:
Thank you for your work - I will note af few observations. Hopefully some of them can be fixed:

1) The version of openwrt is rather old. /etc/banner:
`Based on Openwrt 21.02
 IOT Freedom For End-user!
 HandsomeMod 21.03-Attila-SNAPSHOT`
opkg points to  https://downloads.openwrt.org/releases/21.02-SNAPSHOT/targets/msm89xx... but that cannot be downloaded (not even if I replace https with http)
So It is not possible to get any extra packages
Will the required modifications be added to standard OpenWrt?

2) Only a part of 4GB flash is used:
`root@HandsomeMod:/tmp# df -h
Filesystem                Size      Used Available Use% Mounted on
/dev/root               126.3M    101.3M     22.4M  82% /
tmpfs                   192.6M    592.0K    192.0M   0% /tmp
tmpfs                   512.0K         0    512.0K   0% /dev`

The TX rate of wifi is limited to 6.5Mbs or 13.5Mbs - over ethernet I get >25Mbps download, and >10Mbps download

[image]

3) In Wifi mode Client works ok, but if Client WDS it will nevet get ipadress.
3) In wifimode, the client will be disabled, if the default "SSID: HandsomeMod_A1377C | Mode: Master" is enabled.

[image]

AlienWolfX, 2024-11-06:
> Thank you for your work - I will note af few observations. Hopefully some of them can be fixed

Hello @tboege,

Thank you for your feedback. I'll review the issues that were identified and work on fixing them to the best of my ability.

Best regards, 
Allen

AlienWolfX, 2024-11-06:
> Do you know if it is possible to enter edl or fastboot from your version of OpenWrt (or do I need to short D+ and GND). Maybee I can use reset button?

Yes, for fastboot execute `adb reboot-bootloader` and for edl `fastboot oem reboot-edl`

tboege, 2024-11-06:
Thank you. I was not aware, that those options was available after
installation of openwrt

alexandrglm, 2024-11-06:
> opkg points to https://downloads.openwrt.org/releases/21.02-SNAPSHOT/targets/msm89xx... but that cannot be downloaded (not even if I replace https with http) ...

OpenWRT doesn't officially support the msm8916 chipset, but some Git repositories offer opkg sources for it. These could be integrated into the build system or made available locally.

> In wifimode, the client will be disabled, if the default "SSID: HandsomeMod_A1377C | Mode: Master" is enabled.

As expected with OpenWRT, enabling the default Wi-Fi access point ('SSID: HandsomeMod_A1377C | Mode: Master') disables the client interface.  When radio0 is configured as an AP, the client configuration is disabled, and viceversa.

Hope this helps.

Alexandr.

samthelion17, 2024-11-10:
> but some Git repositories offer opkg sources for it

Where can we find some of those repositories? I just want to install wireguard.

Thanks for all the work you've put into this project! :heart:

AlienWolfX, 2024-11-11:
Hello @samthelion17,

> How can I query the supported bands and other info like imei with mmcli as you wrote in the README?

You can run `mmcli -m 0` to query modem information

> Where can we find some of those repositories? I just want to install wireguard.

> [!NOTE]
I haven't full tested this but [Here's](https://www.mediafire.com/file/gjlysk0ea6nvswc/uz801_with_wireguard.rar/file) a compiled version with wireguard

samthelion17, 2024-11-11:
> I haven't full tested this but [Here's](https://www.mediafire.com/file/gjlysk0ea6nvswc/uz801_with_wireguard.rar/file) a compiled version with wireguard

Can confirm that wireguard is working, and it's working fast! I had to reboot (unplug, re-plug?) after creating and configuring the interface but other than that it works great.

AlienWolfX, 2024-11-11:
> Can confirm that wireguard is working, and it's working fast! 

Thank you for your feedback. ♥️

Best regards,
Allen.

alexandrglm, 2024-11-11:
You may restart with /etc/init.d/the_service restart:
e.g.
# /etc/init.d/network restart