AlienWolfX/UZ801-USB-MODEM #11: HELP!! Change the default IP

tarokeitaro, 2024-11-12:
I'm very new to this, I managed to enter the adb shell, what I need now is how to change the IP segment for example to 192.168.101.0/24?
I'm doing this because I want to connect 5 modems at the same time, unfortunately their default ip is the same and there is no option to change the IP on webui
please help me, thank you.

AlienWolfX, 2024-11-13:
> I'm very new to this, I managed to enter the adb shell, what I need now is how to change the IP segment for example to 192.168.101.0/24?

Hello @tarokeitaro, 

Changing the default IP to another is not straightforward because it requires modifying the MifiService.apk, as the default IP is hardcoded inside. Also incase if you continue on modifying the APK you must also change the ajax url endpoints as it is statically assigned to `192.168.100.1`.

Images for your reference:

Tool used: ***Jadx-GUI***

**GATEWAY**
[image]

**DHCP RANGE**
[image]

tarokeitaro, 2024-11-13:
I have decompiled the MifiService.apk.
then I changed the IP like what you said.
After that I recompiled and signed the app, then installed it to the device. no errors whatsoever.
Then I rebooted, after the Mifi was connected, I still saw the IP still default 192.168.100.1.
where is my mistake?

[image]

[image]

[image]

[image]

AlienWolfX, 2024-11-13:
> I still saw the IP still default 192.168.100.1.\nwhere is my mistake?

I think you need to also change the `initmifiservice.sh` located under `/system/bin`

tarokeitaro, 2024-11-13:
thanks for the answer. it seems to be close to perfect. but is there any configuration left out such as dhcp server? 
I have to configure the ip manually so that it can connect properly.

[image]

[image]

tarokeitaro, 2024-11-13:
I'm sure I've changed it before. I did a mass replacement for all text `192.168.100.` to `192.168.101.` with vscode.

[image]
[image]

AlienWolfX, 2024-11-13:
> I'm sure I've changed it before. I did a mass replacement for all text `192.168.100.` to `192.168.101.` with vscode.

Hello @tarokeitaro,

I see, I guess this is the last thing that you'll need to change under `/system/framework/` you need to modify the `services.jar` values:

[image]

tarokeitaro, 2024-11-13:
I wanted to compile, but got an error as in the picture. previously I decompiled using `jadx`.
sorry because i am very new in java

[image]

AlienWolfX, 2024-11-13:
>  I wanted to compile, but got an error as in the picture. previously I decompiled using `jadx`.
> sorry because i am very new in java

Hello @tarokeitaro,

First and foremost sorry I edited your comment by mistake (LOL).

I'll take a look at this. Also would you mind sending me your services.jar. Thanks.

Best regards,
Allen

tarokeitaro, 2024-11-13:
wow, i'm so grateful. you really helped @AlienWolfX . it worked!!

tarokeitaro, 2024-11-13:
Here's a summary of the many steps I've taken, maybe it will help others who are beginners like me. :)

## How to Change the Default IP Address

### Enable ADB
Go to the following URL to enable ADB
http://192.168.100.1/usbdebug.html

### Download WebUI from Device
```
adb pull /system/priv-app/MifiService.apk
```

### Decompile .apk
```
java -jar apktool.jar d MifiService.apk -o MifiService
```

### Edit IP Address
- Replace all text `192.168.100.` with `192.168.xxx.` as you like.
> Use the Search feature in VSCODE to make things easier.

### Create a test-key
```
git clone https://android.googlesource.com/platform/build
cd build/target/product/security/
openssl pkcs8 -inform DER -nocrypt -in platform.pk8 -out platform.pem
openssl pkcs12 -export -in platform.x509.pem -inkey platform.pem -out platform.p12 -password pass:android -name testkey
keytool -importkeystore -deststorepass android -destkeystore platform.keystore -srckeystore platform.p12 -srcstoretype PKCS12 -srcstorepass android
mv platform.keystore {YOUR_WORK_DIR}
```

### Recompile (passpharase: android)
```
java -jar apktool.jar b -o unsigned.apk MifiService
```

### zipalign
```
zipalign -v 4 unsigned.apk aligned.apk
jarsigner -verbose -sigalg SHA1withRSA -digestalg SHA1 -keystore ./platform.keystore aligned.apk testkey
```

### Install .apk
```
adb install -r aligned.apk
```

### Mount `/system` as RW
```
mount -o rw,remount,rw /system
```

### Edit the file `initmifiservice.sh`
```
busybox vi /system/bin/initmifiservice.sh
```
- Replace the text `192.168.100.` with `192.168.xxx.` as you wish.
> Use the Search feature in VSCODE to make things easier.

> ### If you don't know how to use `vi`
> Type `i`
> Edit the IP you want to assign,
> Click `esc`
> Type `:wq` to write to disk and exit vi

### Download the `services.jar` file
```
adb pull /system/framework/services.jar
```

### Decompile `service.jar`
```
java -jar apktool.jar d -o services services.jar
```

### Edit IP Address
- Replace all text `192.168.100.` with `192.168.xxx.` as you wish.
> Use the Search feature in VSCODE to make it easier.

### Recompile `service.jar`
```
java -jar apktool.jar b -c -f -o services.jar services
```

### Upload `service.jar`
```
adb push services.jar /system/framework/
```

### Set everything back to normal
```
mount -o ro,remount,ro /system
```
> If the system is busy, go to the next step.

### Restart the device
```
adb reboot
```

closed 2024-11-13 by tarokeitaro

AlienWolfX, 2024-11-13:
> wow, i'm so grateful. you really helped @AlienWolfX . it worked!!

Happy to help 😊

Best regards,
Allen