AlienWolfX/UZ801-USB-MODEM #12: Wireguard works iconsistently if I change the settings

samthelion17, 2024-11-14:
So I've been using and testing the OpenWRT firmware with WireGuard that you kindly provided for us (link in #10) and It works... let's just say it works, as long as I don't touch the settings. I can't change the OpenWRT password or the LED configuration, or the theme because it starts acting up... like, I can see that there are sent packages, meaning that it's trying to connect, but zero received. . Maybe it's just a coincidence IDK, still, with the default config there are times that it won't connect, but then I reboot and it works like a charm.

So, is there any way I can help you improve this firmware?

AlienWolfX, 2024-11-14:
> can't change the OpenWRT password or the LED configuration,

Hello @samthelion17,

That's odd what's your dongle board version?.

AlienWolfX, 2024-11-14:
> So, is there any way I can help you improve this firmware?

Hello @samthelion17, [here's](https://github.com/AlienWolfX/HandsomeMod-UZ801) the repository I'm working on.

Some packages makefile such as **backports** and **qcom-firmware** have been modified to work with a local server.

I'll be sending the backports and qcom-firmware files later after I finish my windows update.

Let me know if you need anything

Best regards,
Allen

samthelion17, 2024-11-14:
Thanks! I'll compile and test this weekend if I get the chance.

samthelion17, 2024-11-16:
> I'll be uploading the backports and qcom-firmware files later after I finish my windows update.

Hey I've set up my raspberry pi and left it ready to compile, Just let me know once you've finished uploading the backports and qcom-firmware files. Thanks!

AlienWolfX, 2024-11-17:
Hello @samthelion17,

Sorry for the delay here's the [download link](https://file.io/9KgrkQKJUobF)

closed 2024-11-23 by AlienWolfX