AlienWolfX/UZ801-USB-MODEM #15: I found a new device that can run this version of OpenWRT!

samthelion17, 2024-11-23:
I'm unsure if this is the right place to put it but here it goes:

I found [this device](https://www.aliexpress.us/item/3256806674518599.html) on AliExpress and looking at the specifications I noticed that it uses the same chip as the USB sticks, So I bought it, dumped the stock firmware (adb is enabled from the factory, but the device is not rooted) and immediately flashed [openstick](https://wvthoog.nl/openstick/). To my surprise it didn't work, I was stuck on the bootloader, so I decided to flash your version of OpenWRT and Voilà! A tiny (not so tiny) battery-powered device that can run OpenWRT for $20!

Anyways, If you find this as interesting as I did and want more information about the device just let me know and I'll post it here or somewhere else.
[image]

AlienWolfX, 2024-11-23:
> I found [this device](https://www.aliexpress.us/item/3256806674518599.html) on AliExpress and looking at the specifications I noticed that it uses the same chip as the USB sticks, So I bought it, [...]

Woah, now that's fascinating! If you don't mind removing the cover plate, I'd like to know its hardware specs (RAM, ROM etc). Also  does the modem and WiFi work after flashing OpenWRT?

samthelion17, 2024-11-23:
The WiFi works, the modem... I haven't been able to try it yet 'cause I don't own a SIM card. I will show what's under the shield if it's not soldered but when I checked the build.prop I saw 400something megabytes of RAM, and the firmware dump is about the same size as the original stick so it should have a similar amount of storage. The device runs a version of Android 4.4.4 built on 2024 and the webUI has dramatically improved. Another thing I'd like to try is flash the firmware of this device onto the stick and see if it works.

samthelion17, 2024-11-23:
> > The device runs a version of Android 4.4.4 built on 2024 and the webUI has dramatically improved.
> 
> Interesting, Can you send me the apk file? I'm interested if there's some changes with the backend code as well.
> 
> Thanks.

I will upload the stock firmware folder with the different partitions to the cloud and paste the link here, that way you can study the new UI better. If that wont work for you just let me know and i'll upload only the apk.

Here's the link: Uploading...

AlienWolfX, 2024-11-24:
> I will upload the stock firmware folder with the different partitions to the cloud and paste the link here, that way you can study the new UI better.

Thank you! @samthelion17.

samthelion17, 2024-11-24:
I managed to get openstick working by replacing all .bin files in the 'base' folder with the ones I originally dumped from the device.

closed 2024-11-30 by AlienWolfX