AlienWolfX/UZ801-USB-MODEM #16: Can I turn this into a USB modem without NAT/router

bilogic, 2024-12-21:
Hi,

I bought this device on a whim hoping to use it as an emergency internet connection in case my fibre connection screws up. 
However, plugging it into my router running FreshTomato did not work. And that's how I landed here after some googling.

Getting Debian etc to run on it is sounds like an exciting adventure etc, but for now, I like to ask if there are existing guide/steps to turn it into a pure modem so that my router can remain connected with port forwarding etc all running in an "emergency"?

Thank you, I will join in the rabbit hole when my time allows, meanwhile have fun. :)

AlienWolfX, 2024-12-22:
> I like to ask if there are existing guide/steps to turn it into a pure modem so that my router can remain connected with port forwarding etc all running in an "emergency"?

Hello @bilogic, there is currently no guide available to turn this into a USB modem for routers. However, you can try flashing OpenWRT to your dongle and see if FreshTomato can detect it.

Best Regards,
Allen

bilogic, 2024-12-23:
What? I can flash OpenWRT to my dongle?

Thanks for getting back, I was contemplating flashing OpenWRT instead of FreshTomato, but hesitant as that would be a rabbit hole with my network being down.

AlienWolfX, 2024-12-25:
> What? I can flash OpenWRT to my dongle? 

Yes, you can flash OpenWRT onto that tiny device. Although it is currently in a rough state due to a few bugs here and there, you should give it a shot. You can keep your FreshTomato router as it is.

bilogic, 2024-12-25:
My goal was to have a drop-in fix for the servers in case the fibre connection had issues, which means the fix must provide a public IPv4.

Since the stick gets a private 10.x.x.x IP on the WAN side, I came the to conclusion that there is really no point in turning it into a modem as the servers would still not work.

But I'm going to replace FreshTomato with OpenWRT. Thank you!

closed 2024-12-25 by bilogic