AlienWolfX/UZ801-USB-MODEM #17: Recovering a device that won't boot to EDL using the test points on the board

samthelion17, 2024-12-25:
So I managed to brick one of my devices by flashing the wrong firmware. I tried shorting D+ and GND on the USB port like I always do to boot to EDL but it doesn't work. I tried shorting the two pads on the board that supposedly make the device boot to EDL but it doesn't work either. I'm going to try soldering wires to the pads in case I'm not getting good contact but I'd like to know if there's a way to flash the stock firmware using the tx/rx (is that UART?) pads shown in the picture.
[image]

AlienWolfX, 2024-12-25:
> So I managed to brick one of my devices by flashing the wrong firmware. I tried shorting D+ and GND on the USB port like I always do to boot to EDL but it doesn't work. I tried shorting the two pads on the board that supposedly make the device boot to EDL but it doesn't work either.

What firmware did you flash prior to the failure.?

> I'd like to know if there's a way to flash the stock firmware using the tx/rx (is that UART?)

Yes, that is UART. It's possible to use it, although I haven't tested it yet.

samthelion17, 2024-12-25:
> What firmware did you flash prior to the failure.?

The firmware of the other device that I told you about on another issue. I figured since both can run the same build of OpenWRT and openstick maybe the firmwares were "interchangeable" as well. Guess what, I was wrong. 

> Yes, that is UART. It's possible to use it, although I haven't tested it yet.

I don't have a CH341 programmer or a USB to UART cable but I have a raspberry pi that I can maybe use to interface with the dongle, but how can I force the device into EDL or flash the stock firmware?

AlienWolfX, 2024-12-25:
> The firmware of the other device that I told you about on another issue. I figured since both can run the same build of OpenWRT and openstick maybe the firmwares were \"interchangeable\" as well. Guess what, I was wrong.

When you just plug it into your PC do you see the device under Device Manager.?

samthelion17, 2024-12-25:
I'm using Linux, so I type `dmesg --follow` on the terminal and watch the output and nothing shows up it's like the device is dead. I'm starting to think I killed it while trying to short the pins to force EDL mode, I think I might have shorted something else and now the device is dead because even tho it doesn't show up in the computer it does get warm.
I'm gonna solder some wires to the UART interface and see if the device makes some noise, if it's dead I'll close this issue.

samthelion17, 2024-12-25:
It's toast! I'm not getting anything out of the UART and the processor gets burning hot, so... yeah, I'm giving up on it.

closed 2024-12-25 by samthelion17