AlienWolfX/UZ801-USB-MODEM #18: Building instructions

mbna76, 2025-01-11:
Hello, nice project - got OpenWRT running, but really would like a newer version then 21.02. Do you have the firmware build instructions? Thank you and kind regards

AlienWolfX, 2025-01-12:
Hello build instruction for the version attached can be found here[UZ801 OpenRT](https://github.com/AlienWolfX/HandsomeMod-UZ801)

mbna76, 2025-01-12:
Hello @AlienWolfX so I cloned the repo and followed your instruction on Ubuntu 24.04 LTS. So two comments so far:
1. I needed to add a few lines in ./include/prereq-build.mk because the files was not recognizing my Python 3.12.3 since it was checking only up to 3.10, so I added

python3.12 -V 2>&1 | grep 'Python 3', \
python3.11 -V 2>&1 | grep 'Python 3', \

and 

python3 -V 2>&1 | grep -E 'Python 3\.([5-9]|10|11|12)\.?'))

add insert the number |11|12. This was needed for two prebuild checks.

After this the first two command 
./scripts/feeds update -a
./scripts/feeds install -a
succeeded.

2. But after make menuconfig and at make I ran into the following issue:

make[3]: Entering directory '/home/tlb/HandsomeMod-UZ801/package/kernel/mac80211'
mkdir -p /home/tlb/HandsomeMod-UZ801/dl
SHELL= flock /home/tlb/HandsomeMod-UZ801/tmp/.backports-5.15.33-1.tar.xz.flock -c '  	/home/tlb/HandsomeMod-UZ801/scripts/download.pl "/home/tlb/HandsomeMod-UZ801/dl" "backports-5.15.33-1.tar.xz" "07237fd9f66ce4b95d551569e2f8395342fdb720305f01aa52cd5ca216594278" "" "http://192.168.1.14:8000/"    '
+ curl -f --connect-timeout 20 --retry 5 --location http://192.168.1.14:8000/backports-5.15.33-1.tar.xz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:--  0:00:20 --:--:--     0
curl: (28) Failed to connect to 192.168.1.14 port 8000 after 20002 ms: Timeout was reached
Warning: Problem : timeout. Will retry in 1 seconds. 5 retries left.
  0     0    0     0    0     0      0      0 --:--:--  0:00:20 --:--:--     0
curl: (28) Failed to connect to 192.168.1.14 port 8000 after 20003 ms: Timeout was reached
Warning: Problem : timeout. Will retry in 2 seconds. 4 retries left.
  0     0    0     0    0     0      0      0 --:--:--  0:00:20 --:--:--     0
curl: (28) Failed to connect to 192.168.1.14 port 8000 after 20003 ms: Timeout was reached
Warning: Problem : timeout. Will retry in 4 seconds. 3 retries left.
  0     0    0     0    0     0      0      0 --:--:--  0:00:20 --:--:--     0
curl: (28) Failed to connect to 192.168.1.14 port 8000 after 20003 ms: Timeout was reached
Warning: Problem : timeout. Will retry in 8 seconds. 2 retries left.
  0     0    0     0    0     0      0      0 --:--:--  0:00:20 --:--:--     0
curl: (28) Failed to connect to 192.168.1.14 port 8000 after 20002 ms: Timeout was reached
Warning: Problem : timeout. Will retry in 16 seconds. 1 retries left.
  0     0    0     0    0     0      0      0 --:--:--  0:00:20 --:--:--     0
curl: (28) Failed to connect to 192.168.1.14 port 8000 after 20002 ms: Timeout was reached
Download failed.
+ curl -f --connect-timeout 20 --retry 5 --location https://sources.cdn.openwrt.org/backports-5.15.33-1.tar.xz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 6819k  100 6819k    0     0  3468k      0  0:00:01  0:00:01 --:--:-- 3467k
Hash of the downloaded file does not match (file: 1b6b3bded4c81814ebebe2d194c2f8966d2399005b85ebb0557285b6e73f5422, requested: 07237fd9f66ce4b95d551569e2f8395342fdb720305f01aa52cd5ca216594278) - deleting download.
+ curl -f --connect-timeout 20 --retry 5 --location https://sources.openwrt.org/backports-5.15.33-1.tar.xz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 6819k  100 6819k    0     0  1980k      0  0:00:03  0:00:03 --:--:-- 1980k
Hash of the downloaded file does not match (file: 1b6b3bded4c81814ebebe2d194c2f8966d2399005b85ebb0557285b6e73f5422, requested: 07237fd9f66ce4b95d551569e2f8395342fdb720305f01aa52cd5ca216594278) - deleting download.
+ curl -f --connect-timeout 20 --retry 5 --location https://mirror2.openwrt.org/sources/backports-5.15.33-1.tar.xz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0   153    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
curl: (22) The requested URL returned error: 404
Download failed.
No more mirrors to try - giving up.
make[3]: *** [Makefile:553: /home/tlb/HandsomeMod-UZ801/dl/backports-5.15.33-1.tar.xz] Error 2
make[3]: Leaving directory '/home/tlb/HandsomeMod-UZ801/package/kernel/mac80211'
time: package/kernel/mac80211/regular/compile#0.40#0.51#157.02
    ERROR: package/kernel/mac80211 failed to build (build variant: regular).
make[2]: *** [package/Makefile:116: package/kernel/mac80211/compile] Error 1
make[2]: Leaving directory '/home/tlb/HandsomeMod-UZ801'
make[1]: *** [package/Makefile:110: /home/tlb/HandsomeMod-UZ801/staging_dir/target-mips_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/tlb/HandsomeMod-UZ801'
make: *** [/home/tlb/HandsomeMod-UZ801/include/toplevel.mk:230: world] Error 2

AlienWolfX, 2025-01-13:
> I cloned the repo and followed your instruction on Ubuntu 24.04 LTS.

Hello, I haven't tested building on Ubuntu 24.04. My successful compilation was done on **Ubuntu 20.04**.

closed 2025-01-17 by AlienWolfX