AlienWolfX/UZ801-USB-MODEM #19: Kernel Panic from Corrupted RootFS

udaravima, 2025-01-14:
How can I get to EDL or fastboot from the new bootloader
Mine just looping through a kernel panic
```

[    0.642263] debugfs: Directory '4a9000.thermal-sensor' with parent 'tsens' already present!
[    0.643665] spmi-temp-alarm 200f000.spmi:pmic@0:temp-alarm@2400: failed to register senso▒[    0.932214] Waiting for root device /dev/mmcblk0p14...
[    1.012218] mmc0: new DDR MMC card at address 0001
[    1.013288] mmcblk0: mmc0:0001  XINYH 3.61 GiB
[    1.013750] mmcblk0boot0: mmc0:0001  XINYH partition 1 2.00 MiB
[    1.014219] mmcblk0boot1: mmc0:0001  XINYH partition 2 2.00 MiB
[    1.014379] mmcblk0rpmb: mmc0:0001  XINYH partition 3 2.00 MiB, chardev (245:0)
[    1.017259]  mmcblk0: p1 p2 p3 p4 p5 p6 p7 p8 p9 p10 p11 p12 p13 p14
[    1.052268] List of all partitions:
[    1.052288] b300         3784704 mmcblk0
[    1.052295]  driver: mmcblk
[    1.052302]   b301               2 mmcblk0p1 18285060-b8c8-7cf7-2823-fd5dd2956b88
[    1.052308]
[    1.052313]   b302             512 mmcblk0p2 534641ab-51f1-f296-cf79-26e9c92e9002
[    1.052319]
[    1.052323]   b303             512 mmcblk0p3 4cd3470f-02ef-5e92-c4f4-14bb5251e8f1
[    1.052329]
[    1.052334]   b304            1024 mmcblk0p4 0929ef2f-5cbe-b222-9aff-64578c4e1feb
[    1.052341]
[    1.052345]   b305             512 mmcblk0p5 bf2ea2b6-9f32-b528-99bb-c856cd988976
[    1.052352]
[    1.052356]   b306              16 mmcblk0p6 db68eec7-4c13-bc28-f720-2241bb41d057
[    1.052362]
[    1.052367]   b307            2048 mmcblk0p7 f4c8387d-6628-200b-82cc-16025907d272
[    1.052373]
[    1.052377]   b308            2048 mmcblk0p8 45ba3e2a-d277-68a3-4a11-748d8ef623af
[    1.052383]
[    1.052388]   b309               1 mmcblk0p9 28fa1c81-5b9f-3a57-290b-e8ca46eb0055
[    1.052394]
[    1.052398]   b30a            2048 mmcblk0p10 0d6c74b1-89bd-841e-4b2e-b7b23246967b
[    1.052404]
[    1.052409]   b30b            1024 mmcblk0p11 2432ce91-198e-589b-5d6c-1e2953615a38
[    1.052415]
[    1.052419]   b30c           65536 mmcblk0p12 80780b1d-0fe1-27d3-23e4-9244e62f8c46
[    1.052425]
[    1.052430]   b30d            1024 mmcblk0p13 8b46880a-3de7-53e5-1e74-4602f82e1993
[    1.052436]
[    1.052440]   b30e         3454959 mmcblk0p14 a7ab80e8-e9d1-e8cd-f157-93f69b1d141e
[    1.052446]
[    1.052457] No filesystem could mount root, tried:
[    1.052459]  ext3
[    1.052461]  ext2
[    1.052463]  ext4
[    1.052465]  squashfs
[    1.052467]  vfat
[    1.052469]
[    1.052471] Kernel panic - not syncing:
```

AlienWolfX, 2025-01-14:
> How can I get to EDL or fastboot from the new bootloader

Hello @udaravima you can short the pins below to force your device to edl 

[image]

samthelion17, 2025-01-14:
> How can I get to EDL or fastboot from the new bootloader

If what @AlienWolfX suggested doesn't work for you, try shorting Data+ to GND on the usb port while connecting the dongle, and once connected release them. I've had more luck using this method, and not having to solder the tiny pins is a plus!

udaravima, 2025-01-14:
> If what @AlienWolfX suggested doesn't work for you, try shorting Data+ to GND on the usb port while connecting the dongle, and once connected release them. I've had more luck using this method, and [...]

Actually, I have tried those methods but it's keep looping. Something wrong with the hardware?
EDL pad has around 2.1V .

AlienWolfX, 2025-01-14:
> Actually, I have tried those methods but it's keep looping. Something wrong with the hardware?
> EDL pad has around 2.1V . 

Seems like it has, what is your board version? Also what did you do that resulted in a kernel panic state?

udaravima, 2025-01-14:
> > Actually, I have tried those methods but it's keep looping. Something wrong with the hardware?
> > EDL pad has around 2.1V . 
> 
> Seems like it, what is your board version? Also what did you do that resulted in a kernel panic state?

It is UZ801, I wanted to expand the storage area to 3Gigs, but after I boot into fastboot and erased rootfs. I think my USB connection got loosen while flashing rootfs 🥲

[image]

udaravima, 2025-01-14:
> It is UZ801, I wanted to expand the storage area to 3Gigs, but after I boot into fastboot and erased rootfs. I think my USB connection got loosen while flashing rootfs 🥲
>
> [image]

What are the pads on other side? SPI?
[image]

AlienWolfX, 2025-01-14:
> It is UZ801, I wanted to expand the storage area to 3Gigs, but after I boot into fastboot and erased rootfs. I think my USB connection got loosen while flashing rootfs 🥲

udaravima, 2025-01-14:
> > It is UZ801, I wanted to expand the storage area to 3Gigs, but after I boot into fastboot and erased rootfs. I think my USB connection got loosen while flashing rootfs 🥲
> 
> Uh oh I see, do you get fastboot? Or adb?

I erased it using fastboot, But now I get none

```S - QC_IMAGE_VERSION_STRING=BOOT.BF.3.0.C4.4-00010
S - IMAGE_VARIANT_STRING=HAAAANAZA
S - OEM_IMAGE_VERSION_STRING=wingtech-desktop
S - Boot Config, 0x000002e1
S - Core 0 Frequency, 0 MHz
B -      1545 - PBL, Start
B -      3490 - bootable_media_detect_entry, Start
B -    264521 - bootable_media_detect_success, Start
B -    264525 - elf_loader_entry, Start
B -    266116 - auth_hash_seg_entry, Start
B -    266326 - auth_hash_seg_exit, Start
B -    280684 - elf_segs_hash_verify_entry, Start
B -    339309 - PBL, End
B -    346632 - SBL1, Start
B -    410926 - pm_device_init, Start
D -     14792 - pm_device_init, Delta
B -    426329 - boot_flash_init, Start
D -        30 - boot_flash_init, Delta
B -    430324 - boot_config_data_table_init, Start
D -     29798 - boot_config_data_table_init, Delta - (452 Bytes)
B -    464728 - CDT version:3,Platform ID:24,Major ID:1,Minor ID:0,Subtype:0
B -    471103 - sbl1_ddr_set_params, Start
B -    474732 - cpr_init, Start
D -         0 - cpr_init, Delta
B -    480344 - Pre_DDR_clock_init, Start
D -       183 - Pre_DDR_clock_init, Delta
D -         0 - sbl1_ddr_set_params, Delta
B -    492941 - pm_driver_init, Start
D -      6832 - pm_driver_init, Delta
B -    509380 - clock_init, Start
D -        61 - clock_init, Delta
B -    509441 - Image Load, Start
D -     25773 - QSEE Image Loaded, Delta - (571664 Bytes)
B -    535244 - Image Load, Start
D -       213 - SEC Image Loaded, Delta - (2048 Bytes)
B -    542473 - sbl1_efs_handle_cookies, Start
D -       244 - sbl1_efs_handle_cookies, Delta
B -    550311 - Image Load, Start
D -     17233 - QHEE Image Loaded, Delta - (6808 Bytes)
B -    567574 - Image Load, Start
D -     12597 - RPM Image Loaded, Delta - (145640 Bytes)
B -    580201 - Image Load, Start
D -     20771 - APPSBL Image Loaded, Delta - (224592 Bytes)
B -    601002 - QSEE Execution, Start
D -        61 - QSEE Execution, Delta
B -    606706 - SBL1, End
D -    262422 - SBL1, Delta
S - Flash Throughput, 198000 KB/s  (951204 Bytes,  4788 us)
S - DDR Frequency, 400 MHz
Android Bootloader - UART_DM Initialized!!!
[0] welcome to lk

[0] OpenStick Bootloader V0.01
[0] [Author] HandsomeYingyan <handsomeyingyan@gmail.com>
[0] This Bootloader Has Super Handsome Power!
[0] calling constructors
[0] initializing heap
[0] initializing threads
[0] initializing dpc
[0] initializing timers
[0] creating bootstrap completion thread
[10] top of bootstrap2()
[10] initializing platform
[10] platform_init()
[... 293 lines cut ...]
[    1.051836]   b307            2048 mmcblk0p7 f4c8387d-6628-200b-82cc-16025907d272
[    1.051841] 
[    1.051844]   b308            2048 mmcblk0p8 45ba3e2a-d277-68a3-4a11-748d8ef623af
[    1.051849] 
[    1.051852]   b309               1 mmcblk0p9 28fa1c81-5b9f-3a57-290b-e8ca46eb0055
[    1.051857] 
[    1.051860]   b30a            2048 mmcblk0p10 0d6c74b1-89bd-841e-4b2e-b7b23246967b
[    1.051865] 
[    1.051869]   b30b            1024 mmcblk0p11 2432ce91-198e-589b-5d6c-1e2953615a38
[    1.051873] 
[    1.051877]   b30c           65536 mmcblk0p12 80780b1d-0fe1-27d3-23e4-9244e62f8c46
[    1.051882] 
[    1.051885]   b30d            1024 mmcblk0p13 8b46880a-3de7-53e5-1e74-4602f82e1993
[    1.051890] 
[    1.051893]   b30e         3454959 mmcblk0p14 a7ab80e8-e9d1-e8cd-f157-93f69b1d141e
[    1.051898] 
[    1.051907] No filesystem could mount root, tried: 
[    1.051908]  ext3
[    1.051910]  ext2
[    1.051912]  ext4
[    1.051914]  squashfs
[    1.051915]  vfat
[    1.051917] 
[    1.051918] Kernel panic - not syncing:
```

udaravima, 2025-01-14:
> > It is UZ801, I wanted to expand the storage area to 3Gigs, but after I boot into fastboot and erased rootfs. I think my USB connection got loosen while flashing rootfs 🥲
> 
> Uh oh I see, do you get fastboot? Or adb?
> 
> Also, try Interrupting boot

USB is not even detecting via PC
How can I Interrupt boot. I saw a line to enter failsafe,
But I think it is after the kernel bootup.

AlienWolfX, 2025-01-14:
> How can I Interrupt boot. I saw a line to enter failsafe,

Try pressing the reset button while plugging into the PC

udaravima, 2025-01-14:
> > How can I Interrupt boot. I saw a line to enter failsafe,
> 
> Try pressing the reset button while plugging into the PC

I made it to edl, I cleaned it up and resolder solder joints and a pullup resistor.
Thanks for help @AlienWolfX & @samthelion17

closed 2025-01-15 by AlienWolfX

udaravima, 2025-01-15:
And I was able to expand the rootfs back to 3 Gigs. Thanks ❤️

AlienWolfX, 2025-01-15:
> And I was able to expand the rootfs back to 3 Gigs. Thanks ❤️

@udaravima  If you don't mind what command(s) did you run to expand your rootfs.?

udaravima, 2025-01-15:
I couldn't do it  after bootup, Because we can't unmount rootfs to expand.

- but first I get the ext4 partition from ```simg2img rootfs.img raw_rootfs.img``` 
- Then I created a seperate 3Gig image using ```truncate -s 3G new_partition_name.img```
- copy raw_rootfs.img to new_partition ```dd if=raw_rootfs.img of=new_partition_name.img bs=4M conv=nottrunc```
- attatch new_partition to system as loop drive using ```losetup -f --show new_partition_name.img``` 
- fixed errors on partiton (there were some my case) ```e2fsck -f /dev/loop0```
- then resized the partition ```resize2fs /dev/loop0```
- I ran e2fsck again just to make sure
- then unmount and detach the storage using ```losetup -d /dev/loop0```
- by now I had raw partiton of size 3Gigs but I convert it back to truncate using ```img2simg```