AlienWolfX/UZ801-USB-MODEM #21: OpenWRT not connecting to internet with any sim i inserted in.

redbluexe1, 2025-01-31:
I have tried changing apn settings in the Dashboard in Network interfaces in Handsomemod but it still wouldnt connect. I have tried Two diferent sim that are 3-4 years apart and it still wouldnt work despite windows says there is an internet connection. If you ask for the courier/isps, they are Yes 4G and Maxis Hotlink. The original firmware (i dont have access to it anymore) works with the Maxis sim card. Its just saying "Modem bearer teardown in progress" forever. even then attempted restart.

AlienWolfX, 2025-01-31:
> I have tried changing apn settings in the Dashboard in Network interfaces in Handsomemod but it still wouldnt connect. I have tried Two diferent sim that are 3-4 years apart and it still wouldnt [...]

Do you have the firmware dump of the original firmware?

redbluexe1, 2025-01-31:
no i dont anymore sadly.

AlienWolfX, 2025-01-31:
> no i dont anymore sadly.

What does `mmcli -m 0` show **(please redact serial number and imei)**

redbluexe1, 2025-01-31:
Yes 4G sim (Just noticed it wrote prepaid on the actual sim) :
```
root@HandsomeMod:~# mmcli -m 0
  ----------------------------------
  General  |                   path: /org/freedesktop/ModemManager1/Modem/0
           |              device id: [Umm...]
  ----------------------------------
  Hardware |           manufacturer: 1
           |                  model: 0
           |      firmware revision: UZ801_V3.0_21_V01R01B10  1  [Sep 07 2015 23:00:00]
           |           h/w revision: 10000
           |              supported: gsm-umts, lte
           |                current: gsm-umts, lte
           |           equipment id: [removed]
  ----------------------------------
  System   |                 device: qcom-soc
           |                drivers: rpmsg_chrdev, bam-dmux
           |                 plugin: qcom-soc
           |           primary port: rpmsg0
           |                  ports: rpmsg0 (qmi), rpmsg1 (at), rpmsg_ctrl2 (ignored),
           |                         wwan0 (net), wwan1 (net), wwan2 (net), wwan3 (net), wwan4 (net),
           |                         wwan5 (net), wwan6 (net), wwan7 (net)
  ----------------------------------
  Numbers  |                    own: 601********1
  ----------------------------------
  Status   |                   lock: sim-pin2
           |         unlock retries: sim-pin (3), sim-puk (10), sim-pin2 (3), sim-puk2 (10)
           |                  state: enabled
           |            power state: on
           |            access tech: lte
           |         signal quality: 76% (recent)
  ----------------------------------
  Modes    |              supported: allowed: 3g; preferred: none
           |                         allowed: 4g; preferred: none
           |                         allowed: 3g, 4g; preferred: 4g
           |                         allowed: 3g, 4g; preferred: 3g
           |                current: allowed: 4g; preferred: none
  ----------------------------------
  Bands    |              supported: utran-1, utran-8, eutran-1, eutran-3, eutran-5, eutran-7,
           |                         eutran-8, eutran-20, eutran-38, eutran-40, eutran-41
           |                current: utran-1, utran-8, eutran-1, eutran-3, eutran-5, eutran-7,
           |                         eutran-8, eutran-20, eutran-38, eutran-40, eutran-41
  ----------------------------------
  IP       |              supported: ipv4, ipv6, ipv4v6
  ----------------------------------
  3GPP     |                   imei: [removed]
           |          enabled locks: fixed-dialing
           |            operator id: 50212
           |           registration: idle
  ----------------------------------
  3GPP EPS |   ue mode of operation: csps-1
           |     initial bearer apn: yesnet5
           | initial bearer ip type: ipv4v6
  ----------------------------------
  SIM      |       primary sim path: /org/freedesktop/ModemManager1/SIM/0
           |         sim slot paths: slot 1: /org/freedesktop/ModemManager1/SIM/0 (active)
           |                         slot 2: none
```
Maxis Hotlink 4G sim (it is a maxis sim) :
```
root@HandsomeMod:~# mmcli -m 0
  ----------------------------------
  General  |                   path: /org/freedesktop/ModemManager1/Modem/0
           |              device id: [is this required?]
  ----------------------------------
  Hardware |           manufacturer: 1
           |                  model: 0
           |      firmware revision: UZ801_V3.0_21_V01R01B10  1  [Sep 07 2015 23:00:00]
           |         carrier config: default
           |           h/w revision: 10000
[... 20 lines cut ...]
  Modes    |              supported: allowed: 3g; preferred: none
           |                         allowed: 4g; preferred: none
           |                         allowed: 3g, 4g; preferred: 4g
           |                         allowed: 3g, 4g; preferred: 3g
           |                current: allowed: 4g; preferred: none
  ----------------------------------
  Bands    |              supported: utran-1, utran-8, eutran-1, eutran-3, eutran-5, eutran-7,
           |                         eutran-8, eutran-20, eutran-38, eutran-40, eutran-41
           |                current: utran-1, utran-8, eutran-1, eutran-3, eutran-5, eutran-7,
           |                         eutran-8, eutran-20, eutran-38, eutran-40, eutran-41
  ----------------------------------
  IP       |              supported: ipv4, ipv6, ipv4v6
  ----------------------------------
  3GPP     |                   imei: [no]
           |          enabled locks: fixed-dialing
           |           registration: idle
  ----------------------------------
  3GPP EPS |   ue mode of operation: csps-1
           |     initial bearer apn: yesnet5 <-- supposed to say Maxis or max4g
           | initial bearer ip type: ipv4v6
  ----------------------------------
  SIM      |       primary sim path: /org/freedesktop/ModemManager1/SIM/0
           |         sim slot paths: slot 1: /org/freedesktop/ModemManager1/SIM/0 (active)
           |                         slot 2: none
```

AlienWolfX, 2025-01-31:
> Yes 4G sim (Just noticed it wrote prepaid on the actual sim) :

Can you ping 8.8.8.8 (or any web endpoints) in the openwrt cli?

redbluexe1, 2025-01-31:
both have the same result.
```
root@HandsomeMod:~# ping 8.8.8.8
PING 8.8.8.8 (8.8.8.8): 56 data bytes
ping: sendto: Network unreachable
root@HandsomeMod:~# ping google.com
ping: bad address 'google.com'
root@HandsomeMod:~# ping 8.0.0.8
PING 8.0.0.8 (8.0.0.8): 56 data bytes
ping: sendto: Network unreachable
```

AlienWolfX, 2025-01-31:
> both have the same result.
> ```
> root@HandsomeMod:~# ping 8.8.8.8
> PING 8.8.8.8 (8.8.8.8): 56 data bytes
> ping: sendto: Network unreachable
> root@HandsomeMod:~# ping google.com
> ping: bad address [...]

I see is your SIM prepaid or postpaid?. I think that your ISP is blocking the incoming connection.

`mmcli -m 0 --simple-connect="apn={your_apn}"`

redbluexe1, 2025-01-31:
Yes 4G:
```
root@HandsomeMod:~# mmcli -m 0 --simple-connect="apn=yesnet"
error: couldn't connect the modem: 'Timeout was reached'
root@HandsomeMod:~# mmcli -m 0 --simple-connect="apn=ims"
error: couldn't connect the modem: 'Timeout was reached'
```

Maxis Hotlink (operator id: 50212):
```
root@HandsomeMod:~# mmcli -m 0 --simple-connect="apn=unet"
error: couldn't connect the modem: 'Timeout was reached'
root@HandsomeMod:~# mmcli -m 0 --simple-connect="apn=maxisbb"
error: couldn't connect the modem: 'GDBus.Error:org.freedesktop.ModemManager1.Error.Core.InProgress: Connection request forbidden: operation already in progress'
root@HandsomeMod:~# mmcli -m 0 --simple-connect="apn=maxisbb"
error: couldn't connect the modem: 'Timeout was reached'
root@HandsomeMod:~# mmcli -m 0 --simple-connect="apn=max4g"
error: couldn't connect the modem: 'GDBus.Error:org.freedesktop.ModemManager1.Error.Core.InProgress: Connection request forbidden: operation already in progress'
root@HandsomeMod:~# mmcli -m 0 --simple-connect="apn=max4g"
error: couldn't connect the modem: 'Timeout was reached'
```

Yeah and the `mmcli -m 0` command still says `yesnet5`

AlienWolfX, 2025-02-01:
Try putting the Yes 4G sim card in and reinstall OpenWRT

redbluexe1, 2025-02-01:
How do i reinstall OpenWRT? do i have to use wireless debugging or other ways?

AlienWolfX, 2025-02-01:
> How do i reinstall OpenWRT? do i have to use wireless debugging or other ways?

Same method as with installing OpenWRT from your original firmware

redbluexe1, 2025-02-01:
i used usb debugging but i dont know how to reenable it. connecting to wifi dont work because it keeps restarting after save and apply

AlienWolfX, 2025-02-02:
> i used usb debugging but i dont know how to reenable it. connecting to wifi dont work because it keeps restarting after save and apply

U should run `adb reboot-bootloader` followed by `fastboot oem reboot-edl`

closed 2025-02-12 by AlienWolfX

FrancYescO, 2025-09-24:
basically i was having same issue, and similar command output looks luci have some trouble setting the apn or something like that

in my case seems i've fixed adding in the startup script
`mmcli -m 0 -e && mmcli -m 0 --3gpp-set-initial-eps-bearer-settings='apn=internet.it'` (oviusly set your apn)
and executing
```
uci set network.modem.device='/org/freedesktop/ModemManager1/Modem/0'
uci commit network
```
now the modem connection is up also if it shows no tx/rx
[image]

and this is the output of mmcli
```
oot@HandsomeMod:~# mmcli -m 0
  ----------------------------------
  General  |                   path: /org/freedesktop/ModemManager1/Modem/0
           |              device id: xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
  ----------------------------------
  Hardware |           manufacturer: 1
           |                  model: 0
           |      firmware revision: UZ801_V3.0_21_V01R01B10  1  [Sep 07 2015 23:00:00]
           |         carrier config: default
           |           h/w revision: 10000
           |              supported: gsm-umts, lte
           |                current: gsm-umts, lte
           |           equipment id: 860000000000000
  ----------------------------------
  System   |                 device: qcom-soc
           |                drivers: rpmsg_chrdev, bam-dmux
           |                 plugin: qcom-soc
           |           primary port: rpmsg0
           |                  ports: rpmsg0 (qmi), rpmsg1 (at), rpmsg_ctrl2 (ignored),
           |                         wwan0 (net), wwan1 (net), wwan2 (net), wwan3 (net), wwan4 (net),
           |                         wwan5 (net), wwan6 (net), wwan7 (net)
  ----------------------------------
  Numbers  |                    own: 000000000000
  ----------------------------------
  Status   |                   lock: sim-pin2
           |         unlock retries: sim-pin (3), sim-puk (10), sim-pin2 (3), sim-puk2 (10)
           |                  state: connected
           |            power state: on
           |            access tech: lte
           |         signal quality: 46% (cached)
  ----------------------------------
  Modes    |              supported: allowed: 3g; preferred: none
           |                         allowed: 4g; preferred: none
           |                         allowed: 3g, 4g; preferred: 4g
           |                         allowed: 3g, 4g; preferred: 3g
           |                current: allowed: 3g, 4g; preferred: 3g
  ----------------------------------
  Bands    |              supported: utran-1, utran-8, eutran-1, eutran-3, eutran-5, eutran-7,
           |                         eutran-8, eutran-20, eutran-38, eutran-40, eutran-41
           |                current: utran-1, utran-8, eutran-1, eutran-3, eutran-5, eutran-7,
           |                         eutran-8, eutran-20, eutran-38, eutran-40, eutran-41
  ----------------------------------
  IP       |              supported: ipv4, ipv6, ipv4v6
  ----------------------------------
  3GPP     |                   imei: 8600000000000
           |          enabled locks: fixed-dialing
           |            operator id: 123456
           |          operator name: NAMEOFOPERATOR
           |           registration: home
  ----------------------------------
  3GPP EPS |   ue mode of operation: csps-1
           |    initial bearer path: /org/freedesktop/ModemManager1/Bearer/0
           |     initial bearer apn: internet.it
           | initial bearer ip type: ipv4
  ----------------------------------
  SIM      |       primary sim path: /org/freedesktop/ModemManager1/SIM/0
           |         sim slot paths: slot 1: /org/freedesktop/ModemManager1/SIM/0 (active)
           |                         slot 2: none
  ----------------------------------
  Bearer   |                  paths: /org/freedesktop/ModemManager1/Bearer/1
```