AlienWolfX/UZ801-USB-MODEM #24: State stuck on searching on debian

meownyaaa, 2025-04-28:
Everything on debian works fine except for LTE, when trying to bring it up i get a timeout after 90 seconds and registration/state is stuck on searching. IMEI is intact and my number shows up when using mmcli -m 0. not sure what else to do, i already copied stock modem files to /lib/firmware

[image]

AlienWolfX, 2025-04-28:
> Everything on debian works fine except for LTE, when trying to bring it up i get a timeout after 90 seconds and registration/state is stuck on searching. IMEI is intact and my number shows up when using mmcli -m 0. not sure what else to do, i already copied stock modem files to /lib/firmware

Hello, what's your board version also what is the result of `mmcli -m 0 --enable`

meownyaaa, 2025-04-28:
> Hello, what's your board version also what is the result of `mmcli -m 0 --enable`

Result of "mmcli -m 0 --enable" is 
error: couldn't enable the modem: 'GDBus.Error:org.freedesktop.ModemManager1.Error.Core.Unauthorized: PolicyKit authorization failed: not authorized for 'org.freedesktop.ModemManager1.Device.Control''

Board version is FY_UZ801_V3.2

AlienWolfX, 2025-04-28:
> error: couldn't enable the modem: 'GDBus.Error:org.freedesktop.ModemManager1.Error.Core.Unauthorized: PolicyKit authorization failed: not authorized for 'org.freedesktop.ModemManager1.Device.Control''

Looks like permission issue, are you running as root user?

meownyaaa, 2025-04-28:
> Looks like permission issue, are you running as root user?

I forgot to, after I ran as root I got 'successfully enabled the modem' but the timeout still occurs

meownyaaa, 2025-04-28:
> I see what Debian firmware are you using?

I'm using the "openstick-uz801-v3.0" one from the guide

AlienWolfX, 2025-04-28:
> Here's the files [image]

Give [this](https://github.com/AlienWolfX/UZ801-USB_MODEM/blob/main/rsc/troubleshooting.md#changing-modem-region) a try

meownyaaa, 2025-04-28:
> 
> Have you taken a look at [this](https://github.com/AlienWolfX/UZ801-USB_MODEM/blob/main/rsc/troubleshooting.md#changing-modem-region)

I tried all from common and eu and the same thing is happening

AlienWolfX, 2025-04-28:
> I tried all from common and eu and the same thing is happening

I see. Does that problem occur with a freshly flashed Debian image, without any modifications to `/lib/firmware`?

meownyaaa, 2025-04-28:
> I see. Does that problem occur with a freshly flashed Debian image, without any modifications to `/lib/firmware`?

Yes

meownyaaa, 2025-04-29:
my stick is faulty, the seller was a scammer

closed 2025-04-29 by meownyaaa