AlienWolfX/UZ801-USB-MODEM #26: Activate band 800 in France

gomme600, 2025-04-30:
Hi,

Does anyone know which mbn works for band 20 (800 mhz) ? I have tried multiple and I only have bands 1,3,5,40.

I am running the debian image but even the stock android wouldn't connect with my sim.

Running a network scan shows orange but not sfr who are using band 20.

Thanks

AlienWolfX, 2025-04-30:
Hello @gomme600, what mbn have you tried so far?

gomme600, 2025-05-01:
Hello, I have tried the eu ee mbn and all of the ones in common. I have also tried apac/airtel. The bands never change. I am copying the mbn into /lib/firmware. Is this the correct procedure?

gomme600, 2025-05-01:
user@openstick:~$ sudo qmicli -d /dev/wwan0qmi0 --dms-get-band-capabilities
[/dev/wwan0qmi0] Device band capabilities retrieved:
        Bands: 'wcdma-2100'
        LTE bands: '1, 3, 5, 40'
user@openstick:~$

AlienWolfX, 2025-05-01:
> ```
> openstick:~$ sudo qmicli -d /dev/wwan0qmi0 --dms-get-band-capabilities
> [/dev/wwan0qmi0] Device band capabilities retrieved:
>         Bands: 'wcdma-2100'
>         LTE bands: '1, 3, 5, 40'

For verification, have you executed it as described below?

```bash
   * Upload modem.bin partition to your device (this is from your stock dump created if you have one).
   * SSH or "adb shell" to your device.
   * Mount modem.bin: mount -o loop -t vfat modem.bin /mnt
   * cd /mnt/image/
   * cp modem.*, wcnss.* mba.mbn /lib/firmware/  
   * In modem_pr/mcfg/configs/mcfg_sw/generic/, look for the appropriate mcfg_sw.mbn for your region and copy it to /lib/firmware.
```

gomme600, 2025-05-01:
Yes I have already copied my existing modem partition to /lib/firmware. As the modem didn't get a signal even on stock firmware maybe I have a non EU variant ? Do you have different modem firmware available?

AlienWolfX, 2025-05-01:
> Yes I have already copied my existing modem partition to /lib/firmware. As the modem didn't get a signal even on stock firmware maybe I have a non EU variant ? Do you have different modem firmware available?

If you don't mind, could you send me your modem firmware? I'll review it to cross-reference the firmware versions I currently have, as they might be the same.

gomme600, 2025-05-01:
Here is the full system dump : https://drive.gomme600.ovh/s/AR2x4sDceYgnBew

I am also uploading it to mediafire incase my site goes down : https://www.mediafire.com/file/0arasopzukedcgd/fw.zip/file

Thanks for your help.

AlienWolfX, 2025-05-01:
Hello @gomme600 give this mbn a try: [old_common_default.zip](https://github.com/user-attachments/files/19999126/old_common_default.zip)

Here is also the link for the full modem partition:  https://filebin.net/ec1ujmuzhcvvbbuy

gomme600, 2025-05-01:
Thanks for the firmware! Unfortunately I still only have 4 bands. Here is the mmcli output aswell as a sha256 of the /lib/firmware folder.

```
Linux openstick 6.7.0-rc4-msm8916-g6cc2342c6a72-dirty #1 SMP PREEMPT Sun Jan 14 13:32:37 CET 2024 aarch64

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Thu May  1 13:28:16 2025 from 192.168.200.2
user@openstick:~$ mmcli -m 0
  ----------------------------------
  General  |                   path: /org/freedesktop/ModemManager1/Modem/0
           |              device id: 11fe0c1e7d7c974864f0b5eae78e25f858b97083
  ----------------------------------
  Hardware |           manufacturer: 1
           |                  model: 0
           |      firmware revision: UZ801_V3.0_21_V01R01B10  1  [Sep 07 2015 23:00:00]
           |         carrier config: default
           |           h/w revision: 10000
           |              supported: gsm-umts, lte
           |                current: gsm-umts, lte
           |           equipment id: 869196067636460
  ----------------------------------
  System   |                 device: qcom-soc
           |                drivers: rpmsg_ctrl, qcom-q6v5-mss, bam-dmux
           |                 plugin: qcom-soc
           |           primary port: wwan0qmi0
           |                  ports: rpmsg_ctrl2 (ignored), wwan0 (net), wwan0at0 (at),
           |                         wwan0at1 (at), wwan0qmi0 (qmi), wwan1 (net), wwan2 (net), wwan3 (net),
           |                         wwan4 (net), wwan5 (net), wwan6 (net), wwan7 (net)
  ----------------------------------
  Status   |                   lock: sim-pin2
           |         unlock retries: sim-pin (3), sim-puk (10), sim-pin2 (3), sim-puk2 (10)
           |                  state: enabling
           |            power state: off
  ----------------------------------
  Modes    |              supported: allowed: 3g; preferred: none
           |                         allowed: 4g; preferred: none
           |                         allowed: 3g, 4g; preferred: 4g
           |                         allowed: 3g, 4g; preferred: 3g
           |                current: allowed: 3g, 4g; preferred: 3g
  ----------------------------------
  Bands    |              supported: utran-1, eutran-1, eutran-3, eutran-5, eutran-40
           |                current: utran-1, eutran-1, eutran-3, eutran-5, eutran-40
  ----------------------------------
  IP       |              supported: ipv4, ipv6, ipv4v6
  ----------------------------------
  3GPP     |                   imei: 869196067636460
           |          enabled locks: fixed-dialing
           |   packet service state: detached
  ----------------------------------
  3GPP EPS |   ue mode of operation: csps-1
           | initial bearer ip type: ipv4v6
  ----------------------------------
  SIM      |       primary sim path: /org/freedesktop/ModemManager1/SIM/0
           |         sim slot paths: slot 1: /org/freedesktop/ModemManager1/SIM/0 (active)
           |                         slot 2: /org/freedesktop/ModemManager1/SIM/1
user@openstick:~$ sudo qmicli -d /dev/wwan0qmi0 --dms-get-band-capabilities
[sudo] password for user:
[01 May 2025, 13:32:35] -Warning ** Error reading from istream: Resource temporarily unavailableerror: couldn't create client for the 'dms' service: CID allocation failed in the CTL client: endpoint hangup
user@openstick:~$ sudo qmicli -d /dev/wwan0qmi0 --dms-get-band-capabilities
[/dev/wwan0qmi0] Device band capabilities retrieved:
        Bands: 'wcdma-2100'
        LTE bands: '1, 3, 5, 40'
user@openstick:~$ find /lib/firmware/ -type f -exec sha256sum {} \;
9e37a97571b31b7491fdbb682bbb4b692baea8baed3cd4f257a8a50031ac9d53  /lib/firmware/cmnlib.b01
5d54c3a051bd92c1d2909e9492f8363583e1b995c30d4b74aeac7e04bd36c57d  /lib/firmware/modem.b25
[... 103 lines cut ...]
55f39b1ff316f6e08c9fa2e76575b4e1684c70f0d174af91feee16f4662b9bde  /lib/firmware/modem_pr/mcfg/configs/mcfg_sw/generic/china/ct/svlte/dsda/test/mcfg_sw.mbn
47c43c4ffa34b817086e17f49601eff618d3a0c318a88aa5f3d0221376d400c9  /lib/firmware/modem_pr/mcfg/configs/mcfg_sw/generic/china/ct/svlte/dsda/test_no_/mcfg_sw.mbn
94546832aeb45aa8ecab051c69e1c5e2c51173a810abe1ff562395d88ec0aa46  /lib/firmware/modem_pr/mcfg/configs/mcfg_sw/generic/china/ct/svlte/ss/commerci/mcfg_sw.mbn
2444463ddf140f39dbc188e1652ab074e651ffc8602e580a80b27d70a90569c2  /lib/firmware/modem_pr/mcfg/configs/mcfg_sw/generic/china/ct/cg/dsda/commerci/mcfg_sw.mbn
75bc47df18f3840490ce25b301fccdd566e24ce2d0236276c084687bddcfe237  /lib/firmware/modem_pr/mcfg/configs/mcfg_sw/generic/china/ct/cg/dsds/commerci/mcfg_sw.mbn
08cb573caf7886f727de7f3a6f3aa9d3e4eaede5cb8aa8491475458f29490c79  /lib/firmware/modem_pr/mcfg/configs/mcfg_sw/generic/china/ct/3g/ss/commerci/mcfg_sw.mbn
0f10bec81ec08ad973d391244af0df3811007a7134b753d07428948da14b96a8  /lib/firmware/modem_pr/mcfg/configs/mcfg_sw/generic/china/ct/srte/dsds/cta/mcfg_sw.mbn
a129f4e3abafb4e99a5d410eefc10ea57cad398dc3d36808eb279935ebd6e6ff  /lib/firmware/modem_pr/mcfg/configs/mcfg_sw/generic/china/ct/srlte/ss/commerci/mcfg_sw.mbn
[... cut ...]
59ce78c78a8d0e400503a5c967c19b5db6f4ea008426f9e9aaa1505e74f456c2  /lib/firmware/qcom/venus-1.8/venus.mdt
734cc9af667492724abd934a50244232ae090cd0fe88fad852204a4f06460c48  /lib/firmware/qcom/venus-1.8/venus.b01
c72c50197cfea7b9ffe5933ea616bbaed7246f19c1246fc952608473fbafd2e9  /lib/firmware/qcom/venus-1.8/venus.b00
43b09596e1496809aaa964b34c146580f0566c86bc9d7218a1288fd15fb93424  /lib/firmware/qcom/venus-1.8/venus.b04
8b31eb31ea69c49a5d9193acfc9babdb40ef56b86797ff0b192927ffb7bcf87a  /lib/firmware/qcom/venus-1.8/venus.b03
145d24d815a0159d9b95b8d77b374678f4e99aa0311474d4075aa5728ad6090f  /lib/fir
```

AlienWolfX, 2025-05-01:
> Thanks for the firmware! Unfortunately I still only have 4 bands. Here is the mmcli output aswell as a sha256 of the /lib/firmware folder.

That's strange. On my system, I have more bands using the default `/lib/firmware` and my modem firmware. May I know your board version? Also have you tried reinstalling Debian without modifying the `/lib/firmware`.? If not I would suggest you give it a try as the default alone works internationally.

Best regards,
Allen

gomme600, 2025-05-01:
I didn't originally modify the /lib/firmware after installing debian so the default had the same behavior. I haven't opened the device to see the board revision. I have tried installing your openwrt build and have the same problem.

```
BusyBox v1.33.2 (2022-10-01 04:49:05 UTC) built-in shell (ash)

__________                            __ ____  ___
\______   \ ____   _________ ________/  |\   \/  /
 |    |  _//  _ \ / ___\__  \\_  __ \   __\     /
 |    |   (  <_> ) /_/  > __ \|  | \/|  | /     \
 |______  /\____/\___  (____  /__|   |__|/___/\  \
        \/      /_____/     \/                 \_/
        https://github.com/AlienWolfX
 --------------------------------------------------
 Based on Openwrt 21.02
 IOT Freedom For End-user!
 HandsomeMod 21.03-Attila-SNAPSHOT

=== WARNING! =====================================
There is no root password defined on this device!
Use the "passwd" command to set up a new password
in order to prevent unauthorized SSH logins.
--------------------------------------------------
root@HandsomeMod:~# mmcli
error: no actions specified
root@HandsomeMod:~# mmcli -i 0
  --------------------------------
  General    |               path: /org/freedesktop/ModemManager1/SIM/0
  --------------------------------
  Properties |             active: yes
             |               imsi: 208107598432905
             |              iccid: 89331075240069963390
             | preferred networks: 20810 (gsm, umts)
             |                     26806 (gsm, umts)
             |                     34002 (gsm, umts)
             |                     64710 (gsm, umts)
             |                     37001 (gsm, umts)
             |                     42507 (gsm, umts)
             |                     42528 (gsm, umts)
             |                     21401 (gsm, umts)
             |                     22210 (gsm, umts)
             |                     26202 (gsm, umts)
             |                     23415 (gsm, umts)
             |                     20404 (gsm, umts)
             |                     20205 (gsm, umts)
             |                     22601 (gsm, umts)
             |                     27201 (gsm, umts)
             |                     23003 (gsm, umts)
             |                     21670 (gsm, umts)
             |                     27801 (gsm, umts)
             |                     65501 (gsm, umts)
             |                     27402 (gsm, umts)
             |                     27602 (gsm, umts)
             |                     65101 (gsm, umts)
root@HandsomeMod:~# mmcli -m 0
  ----------------------------------
  General  |                   path: /org/freedesktop/ModemManager1/Modem/0
           |              device id: 11fe0c1e7d7c974864f0b5eae78e25f858b97083
  ----------------------------------
  Hardware |           manufacturer: 1
           |                  model: 0
           |      firmware revision: UZ801_V3.0_21_V01R01B10  1  [Sep 07 2015 23:00:00]
           |           h/w revision: 10000
           |              supported: gsm-umts, lte
           |                current: gsm-umts, lte
           |           equipment id: 869196067636460
  ----------------------------------
  System   |                 device: qcom-soc
           |                drivers: rpmsg_chrdev, bam-dmux
           |                 plugin: qcom-soc
           |           primary port: rpmsg0
           |                  ports: rpmsg0 (qmi), rpmsg1 (at), rpmsg_ctrl2 (ignored),
           |                         wwan0 (net), wwan1 (net), wwan2 (net), wwan3 (net), wwan4 (net),
           |                         wwan5 (net), wwan6 (net), wwan7 (net)
  ----------------------------------
  Status   |                   lock: sim-pin2
           |         unlock retries: sim-pin (3), sim-puk (10), sim-pin2 (3), sim-puk2 (10)
           |                  state: disabled
           |            power state: off
           |         signal quality: 0% (cached)
  ----------------------------------
  Modes    |              supported: allowed: 3g; preferred: none
           |                         allowed: 4g; preferred: none
           |                         allowed: 3g, 4g; preferred: 4g
           |                         allowed: 3g, 4g; preferred: 3g
           |                current: allowed: 3g, 4g; preferred: 3g
  ----------------------------------
  Bands    |              supported: utran-1, eutran-1, eutran-3, eutran-5, eutran-40
           |                current: utran-1, eutran-1, eutran-3, eutran-5, eutran-40
  ----------------------------------
  IP       |              supported: ipv4, ipv6, ipv4v6
  ----------------------------------
  3GPP     |                   imei: 869196067636460
           |          enabled locks: fixed-dialing
  ----------------------------------
  3GPP EPS |   ue mode of operation: csps-1
           | initial bearer ip type: ipv4v6
  ----------------------------------
  SIM      |       primary sim path: /org/freedesktop/ModemManager1/SIM/0
           |         sim slot paths: slot 1: /org/freedesktop/ModemManager1/SIM/0 (active)
           |                         slot 2: /org/freedesktop/ModemManager1/SIM/1
root@HandsomeMod:~# cd /lib/firmware/
root@HandsomeMod:/lib/firmware# ls
mba.mbn        modem.b04      modem.b14      modem.b22      wcnss.b00      wcnss.b10
mcfg_sw.mbn    modem.b05      modem.b15      modem.b23      wcnss.b01      wcnss.b11
modem.b00      modem.b08      modem.b16      modem.b24      wcnss.b02      wcnss.mdt
modem.b01      modem.b10      modem.b17      modem.b25      wcnss.b04      wlan
modem.b02      modem.b11      modem.b18      modem.mdt      wcnss.b06
modem.b03      modem.b13      modem.b19      regulatory.db  wcnss.b09
root@HandsomeMod:/lib/firmware#
```

AlienWolfX, 2025-05-01:
> I didn't originally modify the /lib/firmware after installing debian so the default had the same behavior. I haven't opened the device to see the board revision. I have tried installing your openwrt build and have the same problem.

Is your sim seated properly? Also what does `mmcli -m 0 -e` do?

gomme600, 2025-05-01:
On openwrt I am getting an error enabling the modem:

```
root@HandsomeMod:/lib/firmware# mmcli -m 0 -e
error: couldn't enable the modem: 'GDBus.Error:org.freedesktop.libqmi.Error.Protocol.DeviceNotReady: Couldn't set operating mode: QMI protocol error (52): 'DeviceNotReady''
root@HandsomeMod:/lib/firmware#
``` 

The webui shows : "Modem bearer teardown in progress."

On debian I get this:

mmcli -m 0 -e
error: couldn't enable the modem: 'GDBus.Error:org.freedesktop.ModemManager1.Error.Core.Unauthorized: PolicyKit authorization failed: not authorized for 'org.freedesktop.ModemManager1.Device.Control''
user@openstick:

I am going to try reflashing stock as the modem performed a network scan ok yesterday.

AlienWolfX, 2025-05-01:
> On debian I get this:
> 
> mmcli -m 0 -e
> error: couldn't enable the modem: 'GDBus.Error:org.freedesktop.ModemManager1.Error.Core.Unauthorized: PolicyKit authorization failed: not authorized for 'org.freedesktop.ModemManager1.Device.Control''

You need to run it as `sudo mmcli -m 0 -e` on Debian 

> I am going to try reflashing stock as the modem performed a network scan ok yesterday.

If you're back with stock may you send the output of

`adb shell getprop`

gomme600, 2025-05-01:
I am currently reinstalling stock, I had trouble getting back into edl mode. Adb isn't available in debian and performing adb reboot edl from openwrt just reboots the device normally. I had to short the pads on the pcb. Here is a photo with the board revision.

[image]

gomme600, 2025-05-01:
Here are the properties from stock:

`root@pve:~# adb shell
root@msm8916_32_512:/ # getprop
[DEVICE_PROVISIONED]: [1]
[audio.offload.buffer.size.kb]: [32]
[audio.offload.gapless.enabled]: [true]
[av.offload.enable]: [true]
[config.disable_atlas]: [true]
[dalvik.vm.heapgrowthlimit]: [48m]
[dalvik.vm.heapmaxfree]: [2m]
[dalvik.vm.heapminfree]: [512k]
[dalvik.vm.heapsize]: [128m]
[dalvik.vm.heapstartsize]: [5m]
[dalvik.vm.heaptargetutilization]: [0.75]
[dalvik.vm.lockprof.threshold]: [500]
[dalvik.vm.stack-trace-file]: [/data/anr/traces.txt]
[debug.composition.type]: [c2d]
[debug.egl.hw]: [1]
[debug.force_rtl]: [0]
[debug.mdpcomp.logs]: [0]
[debug.sf.hw]: [1]
[dev.bootcomplete]: [1]
[dev.pm.dyn_samplingrate]: [1]
[dolby.audio.sink.info]: [speaker]
[gsm.apn.sim.operator.numeric]: [20810]
[gsm.current.phone-type]: [1]
[gsm.network.type]: [Unknown]
[gsm.operator.alpha]: []
[gsm.operator.iso-country]: []
[gsm.operator.isroaming]: [false]
[gsm.operator.numeric]: []
[gsm.sim.operator.alpha]: [La Poste Mobile]
[gsm.sim.operator.iso-country]: [fr]
[gsm.sim.operator.numeric]: [20810]
[gsm.sim.state]: [READY]
[gsm.version.baseband]: [N958St_Z85_CN_JSXPH1IDN1H213]
[gsm.version.ril-impl]: [Qualcomm RIL 1.0]
[init.svc.adbd]: [running]
[init.svc.atfwd]: [running]
[init.svc.audiod]: [running]
[init.svc.bootanim]: [stopped]
[init.svc.cnd]: [stopped]
[init.svc.config_bluetooth]: [stopped]
[init.svc.console]: [running]
[init.svc.debuggerd]: [running]
[init.svc.dpmd]: [stopped]
[init.svc.drm]: [running]
[init.svc.fuse_sdcard1]: [running]
[init.svc.fuse_uicc0]: [running]
[init.svc.fuse_usbotg]: [running]
[init.svc.healthd]: [running]
[init.svc.ims_rtp_daemon]: [running]
[init.svc.imsdatadaemon]: [running]
[init.svc.imsqmidaemon]: [running]
[init.svc.initmifiservice]: [stopped]
[init.svc.installd]: [running]
[init.svc.irsc_util]: [stopped]
[init.svc.keystore]: [running]
[init.svc.media]: [running]
[init.svc.mpdecision]: [running]
[init.svc.netd]: [running]
[init.svc.netmgrd]: [running]
[init.svc.ptt_socket_app]: [running]
[init.svc.qcom-c_core-sh]: [stopped]
[init.svc.qcom-c_main-sh]: [stopped]
[init.svc.qcom-post-boot]: [stopped]
[init.svc.qcom-sh]: [stopped]
[init.svc.qcom-usb-sh]: [stopped]
[init.svc.qcomsysd]: [running]
[... 173 lines cut ...]
[ro.telephony.default_network]: [11]
[ro.use_data_netmgrd]: [true]
[ro.vendor.extension_library]: [/vendor/lib/libqc-opt.so]
[ro.wifi.channels]: []
[service.bootanim.exit]: [1]
[swe.tile.height]: [128]
[swe.tile.maxtile]: [128]
[swe.tile.width]: [128]
[sys.boot_completed]: [1]
[sys.ims.DATA_DAEMON_STATUS]: [1]
[sys.ims.QMI_DAEMON_STATUS]: [1]
[sys.sysctl.extra_free_kbytes]: [24300]
[sys.usb.config]: [rndis,serial_smd,diag,adb]
[sys.usb.state]: [rndis,adb]
[telephony.lteOnCdmaDevice]: [1]
[tunnel.audio.encode]: [false]
[usb_uicc.enabled]: [0]
[usb_uicc.loading]: [1]
[use.voice.path.for.pcm.voip]: [true]
[vold.post_fs_data_done]: [1]
[wifi.interface]: [wlan0]
[wlan.driver.ath]: [0]
[wlan.driver.config]: [/data/misc/wifi/WCNSS_qcom_cfg.ini]``

```

AlienWolfX, 2025-05-01:
Looks good, can you access the internet with stock?

gomme600, 2025-05-01:
Unfortunately not, I don't have a signal

adb shell dumpsys telephony.registry | grep -i signalstrength
  mSignalStrength=SignalStrength: 99 0 -120 -160 -120 -1 -1 99 2147483647 2147483647 2147483647 2147483647 2147483647 gsm|lte

AlienWolfX, 2025-05-01:
> Unfortunately not, I don't have a signal
> 
> adb shell dumpsys telephony.registry | grep -i signalstrength
>   mSignalStrength=SignalStrength: 99 0 -120 -160 -120 -1 -1 99 2147483647 2147483647 2147483647 2147483647 2147483647 gsm|lte

Does it show unknown when accessing the webpage at 192.168.100.1?,

gomme600, 2025-05-02:
Success! I was able to enable all of the 4g bands using qpst and qxdm (I had to set the system date to 2018 for installation because the license has expired). Service Programming is only for 3g bands from what I read. After looking on various websites I found screenshots showing the same 4 bands from other people so it appears that there is a version that only has these 4 bands enabled on stock. Thanks for all of your help!

AlienWolfX, 2025-05-02:
> Success! I was able to enable all of the 4g bands using qpst and qxdm (I had to set the system date to 2018 for installation because the license has expired). Service Programming is only for 3g [...]

You're welcome, May I know what settings you modified in QXDM for this to work.

AlienWolfX, 2025-05-04:
> Thanks for the instructions. I'm trying to follow them but the device doesn't appear in QPST. Is there a step to put it in some sort of debug mode?

You need to access `http://192.168.100.1/ms.html` and enable

gomme600, 2025-05-04:
@AlienWolfX I followed the quick start guide here : https://xdaforums.com/t/guide-add-all-gsm-and-lte-bands-to-your-phone.2886059/

gomme600, 2025-05-04:
@ant-thomas You have to install the qpst driver aswell: https://qcomdriver.com/
My device was detected after that. It is rooted and has adb enabled.
I ran this command aswell but I don't know if it was necessary as it was already detected before:

su
setprop sys.usb.config diag,adb

ant-thomas, 2025-05-04:
Thanks both @gomme600 @AlienWolfX 
Those steps got it going - now to test if B20 is available.

Do either of you know what the antenna port type is?
Doesn't look like the standard ufl/ipex I've seen before.

[image]

AlienWolfX, 2025-05-04:
> Do either of you know what the antenna port type is? Doesn't look like the standard ufl/ipex I've seen before.

I think it's this description says `IPEX U.FL UFL Female SMD SMT Solder`

https://www.amazon.in/GraspaDeal-Female-Solder-Coaxial-Connector/dp/B09XF355Y1

ant-thomas, 2025-05-04:
> Success! I was able to enable all of the 4g bands using qpst and qxdm (I had to set the system date to 2018 for installation because the license has expired). Service Programming is only for 3g [...]

Did you write `17592185995263` to NV 06828 & 06829 or did you calculate a specific set of bands?

AlienWolfX, 2025-05-12:
> I've written `3` to NV6828 to enable bands 1 and 2 since I'm in the US and my carrier uses band 2. 

You need to put something like this `17592185995263` to unlock the whole bands

WeirdTreeThing, 2025-05-12:
I also tried that and had the exact same issue.

danyaPostfactum, 2025-10-21:
>LTE bands: '1, 3, 5, 40'

In my country the required set is 1, 3, 7, 8, 20, 38, 40. The set i can test in my area is 1, 3, 7, 38. Unfortunately, modem works with 1 and 3 LTE BANDS in my area only. This is very bad. I need 38 at least.

6828(LTE_BC_CONFIG) NV ITEM defines supported LTE BANDs by the device. The stock value (not sure) is _1786706919637_ (1,3,5,7,8,20,38,40,41). But the BAND 7 and the BAND 38 currentrly are not really supported by UZ801 (and UFI003).

How can I fix it?

As far as I undestand, true band support depends on:
* hardware (what exactrly should we check? MSM8916 LTE basic support is good)
* NV items values or EFS files. (what items exactrly? can we adopt these from other MSM8916 devices?)
* modem firmware (MBN?)

Any ideas and thougths?

Currently I use stock firmware (android). How can i play with MBN now?
I think I should try openstick again (later).