AlienWolfX/UZ801-USB-MODEM #28: Modem goes offline on openwrt

gomme600, 2025-05-06:
On openwrt the modem seems to disappear after some time (around 12 hours).
mmcli can't see the modem anymore:

```
mmcli -m 0
error: couldn't find modem
``` 

After restarting modemmanager it seems to come back (with a few minutes delay) but was disabled.

After running mmcli -m 0 -e it reconnected.

AlienWolfX, 2025-05-06:
> On openwrt the modem seems to disappear after some time (around 12 hours).
> mmcli can't see the modem anymore:
>
> ```
> mmcli -m 0
> error: couldn't find modem
> ```
>
> After restarting modemmanager it seems [...]

Can you send me the output of `dmesg` when this problem occurs

gomme600, 2025-05-07:
Here is the outout:

```
root@HandsomeMod:~# dmesg
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd030]
[    0.000000] Linux version 5.10.33 (ubuntu@allenxemma) (aarch64-openwrt-linux-musl-gcc (OpenWrt GCC 8.4.0 r0-fa7f0be) 8.4.0, GNU ld (GNU Binutils) 2.34) #0 SMP Sat Oct 1 04:49:05 2022
[    0.000000] Machine model: Handsome OpenStick UZ801
[    0.000000] earlycon: msm_serial_dm0 at MMIO 0x00000000078b0000 (options '')
[    0.000000] printk: bootconsole [msm_serial_dm0] enabled
[    0.000000] cma: Reserved 32 MiB at 0x000000009e000000
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000080000000-0x000000009fffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000080000000-0x0000000085ffffff]
[    0.000000]   node   0: [mem 0x0000000086000000-0x000000008bcfffff]
[    0.000000]   node   0: [mem 0x000000008bd00000-0x000000008defffff]
[    0.000000]   node   0: [mem 0x000000008df00000-0x000000008eafffff]
[    0.000000]   node   0: [mem 0x000000008eb00000-0x000000009fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000080000000-0x000000009fffffff]
[    0.000000] On node 0 totalpages: 131072
[    0.000000]   DMA zone: 2048 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 131072 pages, LIFO batch:31
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.0 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.0
[    0.000000] psci: OSI mode supported.
[    0.000000] percpu: Embedded 18 pages/cpu s33304 r8192 d32232 u73728
[    0.000000] pcpu-alloc: s33304 r8192 d32232 u73728 alloc=18*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 129024
[    0.000000] Kernel command line: earlycon console=tty0 console=ttyMSM0,115200 root=/dev/mmcblk0p14 rw rootwait androidboot.emmc=true androidboot.serialno=1a3dbe8 androidboot.baseband=msm mdss_mdp.panel=0:dsi:0:
[    0.000000] Dentry cache hash table entries: 65536 (order: 7, 524288 bytes, linear)
[    0.000000] Inode-cache hash table entries: 32768 (order: 6, 262144 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 360244K/524288K available (8190K kernel code, 980K rwdata, 2332K rodata, 1472K init, 295K bss, 131276K reserved, 32768K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu: 	CONFIG_RCU_FANOUT set to non-default value of 32.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] random: get_random_bytes called from start_kernel+0x348/0x514 with crng_init=0
[    0.000000] arch_timer: cp15 and mmio timer(s) running at 19.20MHz (phys/virt).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000000] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.010830] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.018748] pid_max: default: 32768 minimum: 301
[    0.029295] Mount-cache hash table entries: 1024 (order: 1, 8192 bytes, linear)
[    0.033936] Mountpoint-cache hash table entries: 1024 (order: 1, 8192 bytes, linear)
[    0.042520] rcu: Hierarchical SRCU implementation.
[    0.049086] dyndbg: Ignore empty _ddebug table in a CONFIG_DYNAMIC_DEBUG_CORE build
[    0.054047] smp: Bringing up secondary CPUs ...
[    0.061631] Detected VIPT I-cache on CPU1
[    0.065661] CPU1: Booted secondary processor 0x0000000001 [0x410fd030]
[    0.070411] Detected VIPT I-cache on CPU2
[    0.076243] CPU2: Booted secondary processor 0x0000000002 [0x410fd030]
[    0.080870] Detected VIPT I-cache on CPU3
[    0.086742] CPU3: Booted secondary processor 0x0000000003 [0x410fd030]
[    0.090866] smp: Brought up 1 node, 4 CPUs
[    0.097226] SMP: Total of 4 processors activated.
[    0.101291] CPU features: detected: 32-bit EL0 Support
[    0.106066] CPU features: detected: CRC32 instructions
[... 256 lines cut ...]
[   41.392012] br-lan: port 1(usb0) entered blocking state
[   41.392027] br-lan: port 1(usb0) entered forwarding state
[   41.933043] br-lan: port 2(wlan0) entered blocking state
[   41.933062] br-lan: port 2(wlan0) entered disabled state
[   41.933337] device wlan0 entered promiscuous mode
[   41.933453] br-lan: port 2(wlan0) entered blocking state
[   41.933466] br-lan: port 2(wlan0) entered forwarding state
[   41.941636] wcn36xx: ERROR HAL_8023_MULTICAST_LIST rsp failed err=16
[   41.943843] wcn36xx: ERROR HAL_8023_MULTICAST_LIST rsp failed err=16
[   42.436760] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[   42.437505] br-lan: port 2(wlan0) entered disabled state
[   42.801320] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   42.801522] br-lan: port 2(wlan0) entered blocking state
[... cut ...]
[ 3650.860530] remoteproc remoteproc0: recovering 4080000.remoteproc
[ 3652.660725] qcom-wcnss-pil a204000.remoteproc: unexpected response to sysmon event
[ 3652.660741] remoteproc remoteproc0: stopped remote processor 4080000.remoteproc
[ 3652.661682] qcom-wcnss-pil a204000.remoteproc: unexpected response to sysmon event
[ 3652.720218] qcom-q6v5-mss 4080000.remoteproc: MBA booted without debug policy, loading mpss
[ 3653.361310] remoteproc remoteproc0: remote processor 4080000.remoteproc is now up
[ 3656.830694] bam-dmux wwan: Failed to resume: -22
```