AlienWolfX/UZ801-USB-MODEM #31: Disable Wi-Fi tethering

julien31-c, 2025-07-05:
Hi,
I'm trying to disable Wi-Fi tethering but keep USB tethering enabled. The modem is still running Android, I've rooted it and I'm trying to turn it off via ADB, but it automatically turns back on. Do you have any suggestions? Thanks in advance.

AlienWolfX, 2025-07-05:
> Hi, I'm trying to disable Wi-Fi tethering but keep USB tethering enabled. The modem is still running Android, I've rooted it and I'm trying to turn it off via ADB, but it automatically turns back on.

Hello @julien31-c, yes, Wi-Fi tethering is indeed managed by `MifiService.apk`, along with the DHCP and carrier/SIM-related functionality.

julien31-c, 2025-07-06:
It's on the ISP side. With the stock version (android and without killing MifiService.apk) the IPv4 assignment works without any issues.

```
user@openstick:~$ sudo mmcli -m 0
  ----------------------------------
  General  |                   path: /org/freedesktop/ModemManager1/Modem/0
           |              device id: b170d6b789669b5d8c294a5925d935e4c3c3b287
  ----------------------------------
  Hardware |           manufacturer: 1
           |                  model: 0
           |      firmware revision: UZ801_V3.0_21_V01R01B10  1  [Sep 07 2015 23:00:00]
           |         carrier config: default
           |           h/w revision: 10000
           |              supported: gsm-umts, lte
           |                current: gsm-umts, lte
           |           equipment id: 8600XXXXXXXXXXX
  ----------------------------------
  System   |                 device: qcom-soc
           |                drivers: rpmsg_ctrl, qcom-q6v5-mss, bam-dmux
           |                 plugin: qcom-soc
           |           primary port: wwan0qmi0
           |                  ports: rpmsg_ctrl2 (ignored), wwan0 (net), wwan0at0 (at),
           |                         wwan0at1 (at), wwan0qmi0 (qmi), wwan1 (net), wwan2 (net), wwan3 (net),
           |                         wwan4 (net), wwan5 (net), wwan6 (net), wwan7 (net)
  ----------------------------------
  Status   |                   lock: sim-pin2
           |         unlock retries: sim-pin (3), sim-puk (10), sim-pin2 (3), sim-puk2 (10)
           |                  state: connected
           |            power state: on
           |            access tech: lte
           |         signal quality: 54% (recent)
  ----------------------------------
  Modes    |              supported: allowed: 3g; preferred: none
           |                         allowed: 4g; preferred: none
           |                         allowed: 3g, 4g; preferred: 4g
           |                         allowed: 3g, 4g; preferred: 3g
           |                current: allowed: 4g; preferred: none
  ----------------------------------
  Bands    |              supported: utran-1, utran-8, eutran-1, eutran-3, eutran-5, eutran-7,
           |                         eutran-8, eutran-20, eutran-38, eutran-40, eutran-41
           |                current: utran-1, utran-8, eutran-1, eutran-3, eutran-5, eutran-7,
           |                         eutran-8, eutran-20, eutran-38, eutran-40, eutran-41
  ----------------------------------
  IP       |              supported: ipv4, ipv6, ipv4v6
  ----------------------------------
  3GPP     |                   imei: 8600XXXXXXXXXXX
           |          enabled locks: fixed-dialing
           |            operator id: 20820
           |          operator name: Mobile
           |           registration: home
           |   packet service state: attached
           |                    pco: 0: (complete) '27078000000300F000'

  ----------------------------------
  3GPP EPS |   ue mode of operation: csps-1
           |    initial bearer path: /org/freedesktop/ModemManager1/Bearer/0
           |     initial bearer apn: mmsbouygtel.com
           | initial bearer ip type: ipv4v6
  ----------------------------------
  SIM      |       primary sim path: /org/freedesktop/ModemManager1/SIM/0
           |         sim slot paths: slot 1: /org/freedesktop/ModemManager1/SIM/0 (active)
           |                         slot 2: none
  ----------------------------------
  Bearer   |                  paths: /org/freedesktop/ModemManager1/Bearer/1
```

```
user@openstick:~$ ip a show wwan0
4: wwan0: <POINTOPOINT,NOARP,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UNKNOWN group default qlen 1000
    link/[519]
[... 105 lines cut ...]
proxy.method:                           none
proxy.browser-only:                     no
proxy.pac-url:                          --
proxy.pac-script:                       --
GENERAL.NAME:                           lte
GENERAL.UUID:                           270bfed9-7e8e-4b33-b5f8-4ffbdcb4dc2e
GENERAL.DEVICES:                        wwan0qmi0
GENERAL.IP-IFACE:                       wwan0
GENERAL.STATE:                          activated
GENERAL.DEFAULT:                        no
GENERAL.DEFAULT6:                       yes
GENERAL.SPEC-OBJECT:                    --
GENERAL.VPN:                            no
GENERAL.DBUS-PATH:                      /org/freedesktop/NetworkManager/ActiveConnection/8
GENERAL.CON-PATH:                       /org/freedesktop/NetworkManager/Settings/1
GENERAL.ZONE:                           --
GENERAL.MASTER-PATH:                    --
IP4.GATEWAY:                            --
IP6.ADDRESS[1]:                         2a04:cec0:11ea:bf51:f9e5:f84c:2efb:d702/64
IP6.GATEWAY:                            2a04:cec0:11ea:bf51:c5ab:4af9:98e1:f0a8
IP6.ROUTE[1]:                           dst = 2a04:cec0:11ea:bf51::/64, nh = ::, mt = 700
IP6.ROUTE[2]:                           dst = ::/0, nh = 2a04:cec0:11ea:bf51:c5ab:4af9:98e1:f0a8, mt = 700
IP6.DNS[1]:                             2001:860:d002:1201::129:202
IP6.DNS[2]:                             2001:860:d002:1201::129:201
```

AlienWolfX, 2025-07-07:
Try forcing mmcli to use ipv4 : 
```bash
sudo nmcli connection modify lte ipv4.method auto ipv6.method ignore
sudo nmcli connection up lte
```

julien31-c, 2025-07-07:
`Jul 07 21:49:57 openstick NetworkManager[244]: <warn>  [1751917797.8664] device (wwan0qmi0): retrieving IP configuration failed: modem IP method unsupported`
`Jul 07 21:50:54 openstick NetworkManager[244]: <warn>  [1751917854.3414] modem-broadband[wwan0qmi0]: failed to connect modem: Call failed: internal error: pdn-ipv4-call-disallowed>`

I have this error message in the sysctl logs. The ISP seems not to allow IPV4 for my device. I thought it was due to the IMEI but with Android it works and the IMEI is identical. I figure that with Android it considers me as a mobile phone. The MifiService apk must handle something that I don't have with Debian.

AlienWolfX, 2025-07-08:
> `Jul 07 21:49:57 openstick NetworkManager[244]: <warn>  [1751917797.8664] device (wwan0qmi0): retrieving IP configuration failed: modem IP method unsupported`
> `Jul 07 21:50:54 openstick [...]

It kinda looks like your ISP doesn't support IPv4 address but I might be wrong did you try it with another device? Preferably a Phone.

AlienWolfX, 2025-07-08:
> `Jul 07 21:49:57 openstick NetworkManager[244]: <warn>  [1751917797.8664] device (wwan0qmi0): retrieving IP configuration failed: modem IP method unsupported`
> `Jul 07 21:50:54 openstick [...]

Maybe the UZ801 modem firmware is too outdated for your ISP, did you try flashing your modem partition?

AlienWolfX, 2025-07-08:
> I reflashed the stock version and started modifying the MifiService APK. I'm making my changes in the smali file \"WifiApController\", then I rebuild, align, sign, and install it using adb push. I don't get any error messages. However, my modifications aren't reflected... I installed your version of the APK, and everything works as expected. I'm clearly missing something in the process, even though there are no errors.

Copy I'll take a look at it

julien31-c, 2025-07-08:
> Copy I'll take a look at it

https://github.com/julien31-c/apk_uz_801.git
 
Thank you.

julien31-c, 2025-07-09:
Alongside trying to disable the WiFi hotspot, I’m attempting to change the gateway IP from 192.168.100.1 to 192.169.101.1 so that I can use two UZ801 devices on the same PC.

I changed the IP values on both the front-end and back-end sides within MifiService. I also modified the mificonfiguration.json file in /data as well as initmifiservice.sh in /system/bin. I managed to change the gateway, but the MifiService is acting up — it keeps restarting the USB tethering. Meanwhile, the front-end communicates properly with the back-end. Everything seems to work, and the local IP is correctly within the new DHCP range.

AlienWolfX, 2025-07-10:
> I changed the IP values on both the front-end and back-end sides within MifiService. I also modified the mificonfiguration.json file in /data as well as initmifiservice.sh in /system/bin. I managed [...]

You're missing this step [Modifying services.jar](https://github.com/AlienWolfX/UZ801-USB_MODEM/issues/11#issuecomment-2473078900)

julien31-c, 2025-07-10:
> > I changed the IP values on both the front-end and back-end sides within MifiService. I also modified the mificonfiguration.json file in /data as well as initmifiservice.sh in /system/bin. I managed to change the gateway, but the MifiService is acting up — it keeps restarting the USB tethering. Meanwhile, the front-end communicates properly with the back-end. Everything seems to work, and the local IP is correctly within the new DHCP range.
> 
> I think you're missing this step [Modifying services.jar](https://github.com/AlienWolfX/UZ801-USB_MODEM/issues/11#issuecomment-2473078900)

Great, that was it — thank you so much!

I’ve also found some answers regarding why I wasn’t getting an IPv4 address with Debian. I tested with another internet provider, and it worked fine.

I think it’s because the first ISP uses an IPv6-only network, with IPv4 handled through NAT64 + DNS64 or 464XLAT. On Debian, this probably isn’t managed natively, whereas on Android, it's likely handled by default.

danyaPostfactum, 2025-08-22:
It looks like MifiService just calls android/net/wifi/WifiManager;->setWifiApEnabled. Then some script creates a bringe betwen wlan0 and rndis0. So, if we want to disable wifi hotspot, we need to somehow manually start usb tethering. Here is filtered logcat of setWifiApEnabled:
```
WifiApController         D  setWifiApConfig 4G-UFI-DA3 3 123456789
WifiApController         D  *** setWifiApEnabled CALLED **** true
WifiService              D  setWifiEnabled: false pid=1659, uid=1000
Tethering                D  sendTetherStateChangedBroadcast 0, 0, 0
Tethering                D  InitialState.processMessage what=4
Tethering                D  sendTetherStateChangedBroadcast 0, 0, 0
Tethering                D  WIFI_AP_STATE_CHANGED: wifiApState=12
SoftapController         D  Softap fwReload - Ok
SoftapController         D  SoftAP started successfully
hostapd                  E  Configuration file: /data/misc/wifi/hostapd.conf
hostapd                  E  MAC list file '/data/hostapd/hostapd.accept' not found.
hostapd                  E  Line 173: Failed to read accept_mac_file '/data/hostapd/hostapd.accept'
hostapd                  E  MAC list file '/data/hostapd/hostapd.deny' not found.
hostapd                  E  Line 174: Failed to read deny_mac_file '/data/hostapd/hostapd.deny'
hostapd                  I  rfkill: Cannot open RFKILL control device
hostapd                  E  Using interface wlan0 with hwaddr a0:08:40:86:5d:a3 and ssid "4G-UFI-DA3"
Tethering                D  sendTetherStateChangedBroadcast 1, 0, 0
Tethering                D  WIFI_AP_STATE_CHANGED: wifiApState=13
Tethering                D  Tethering wlan0
Tethering                D  InitialState.processMessage what=2
Tethering                D  sendTetherStateChangedBroadcast 0, 0, 0
TetherController         D  tetherInterface(br0)
TetherController         D  addV6RtrAdvIface: len = 1. Iface: br0
TetherController         D  Router advertisement daemon already stopped
TetherController         D  Router advertisement daemon running
radish                   D  radish_bridge_debug_cmds: ebtables -t broute -L
Tethering                D  Tethered wlan0
Tethering                D  sendTetherStateChangedBroadcast 0, 1, 0
radish                   D  Bridge table: broute
radish                   D  
radish                   D  Bridge chain: BROUTING, entries: 0, policy: ACCEPT
radish                   D  radish_bridge_debug_cmds: brctl show
Tethering                D  sendTetherStateChangedBroadcast 1, 1, 0
Tethering                D  MasterInitialState.processMessage what=1
radish                   D  bridge name	bridge id		STP enabled	interfaces
radish                   D  br0		8000.7ef10a7f0a3d	no		rndis0
radish                   D  							wlan0
radish                   D  radish_bridge_debug_cmds: netstat -ap
TetherController         D  Setting IP forward enable = 1
TetherController         D  config_write_setting(/proc/sys/net/ipv6/conf/all/proxy_ndp, 2)
TetherController         D  config_write_setting(/proc/sys/net/ipv6/conf/all/forwarding, 2)
radish                   D  Proto Recv-Q Send-Q Local Address          Foreign Address        State
radish                   D   tcp       0      0 0.0.0.0:7628           0.0.0.0:*              LISTEN
radish                   D   tcp       0      0 127.0.0.1:5037         0.0.0.0:*              LISTEN
radish                   D  radish_init_bridge: brctl addbr br0
TetherController         D  Starting tethering services
radish                   E  Cannot create the bridge interface
radish                   D  radish_init_bridge: ifconfig br0 up
TetherController         D  Sending update msg to dnsmasq [update_ifaces:br0]
TetherController         D  Tethering services running
radish                   D  Adding br0 to br0
radish                   D  radish_parse_args: brctl addif br0 br0
radish                   E  Unable to add the interface to the bridge
TetherController         D  setDnsForwarders(0 = '8.8.8.8')
TetherController         D  setDnsForwarders(1 = '8.8.4.4')
TetherController         D  Sending update msg to dnsmasq [update_dns:8.8.8.8:8.8.4.4]
Tethering                D  chooseUpstreamType(true), preferredApn =5, got type=0
Tethering                I  Finding IPv4 upstream interface on: {InterfaceName: rmnet0 LinkAddresses: /29,]  Routes: [0.0.0.0/0 -> 10.2.195.213,] DnsAddresses: [172.30.0.73,172.30.0.74,] Domains: }
Tethering                I  Found interface rmnet0
radish                   E   FATAL: Need to specify an interface to proxy
radish                   D  Interfaces found: 
radish                   D  br0
radish                   E  Need at least 2 interfaces
radish                   D  radish_cleanup_handler: ip -6 route flush dev br0
TetherController         D  setDnsForwarders(0 = '172.30.0.73')
TetherController         D  setDnsForwarders(1 = '172.30.0.74')
TetherController         D  Sending update msg to dnsmasq [update_dns:172.30.0.73:172.30.0.74]
Tethering                D  notifying tethered with iface =rmnet0
Tethering                D  TetheredState.processMessage what=12
NatController            V  enableNat(intIface=<br0>, extIface=<rmnet0>)
NatController            V  runCmd(/system/bin/iptables -t nat -A natctrl_nat_POSTROUTING -o rmnet0 ) res=0
NatController            V  runCmd(/system/bin/iptables -A natctrl_FORWARD -i rmnet0 -o br0 -m state LISHED,RELATED -g natctrl_tether_counters) res=0
[... cut ...]
NatController            V  runCmd(/system/bin/iptables -A natctrl_FORWARD -i br0 -o rmnet0 -g r_counters) res=0
NatController            V  runCmd(/system/bin/iptables -A natctrl_tether_counters -i br0 -o rmnet0 ame br0_rmnet0 --grow -j RETURN) res=0
NatController            V  runCmd(/system/bin/iptables -A natctrl_tether_counters -i rmnet0 -o br0 ame rmnet0_br0 --grow -j RETURN) res=0
NatController            V  runCmd(/system/bin/iptables -D natctrl_FORWARD -j DROP) res=0
NatController            V  runCmd(/system/bin/iptables -A natctrl_FORWARD -j DROP) res=0
```