AlienWolfX/UZ801-USB-MODEM #33: Unable to install Software

nalz0, 2025-08-22:
Hello!

First up, I just want to say what an amazing project this is! I'm so glad OpenWRT is able to install on UZ801 because the stock firmware is absolutely awful.

I've noticed I can't install any software, most likely due to incompatibility with MSM8916 as I saw in a different post.

I was just wondering if there was any possibility of getting Travelmate on this? It's a software that allows to connect to other networks and repeats them, even if they have captive portals. I understand there may be a manual way of setting this configuration up but I do not have any knowledge whatsoever with OpenWRT when it comes to configuring it. It's actually my first time using it!

Thanks.

nalz0, 2025-08-24:
Hello!

Thanks for your response. Could you help me out with building Travelmate? I've never built software from the source ever and I do not have any idea on how to go about doing it.

Thanks.

AlienWolfX, 2026-03-05:
Build(With TravelMate): [OpenWRT-UZ801_2026](https://github.com/AlienWolfX/HandsomeMod-UZ801/releases/download/v1.0-alpha/OpenWRT-UZ801_2026.tar.xz)